mongoose-os-apps/shelly-homekit #1280: No automation in homekit

heili1094, 2023-05-18:
Hi, since update to new homekit architecture the shelly 1 with latest firmware will not appear if i add new automation with sensor trigger. 
kindly help! I use the shelly as rain sensor.
Thx

timoschilling, 2023-05-18:
Please post a screenshot of your Shelly config

heili1094, 2023-05-18:
[image]
[image]
[image]

heili1094, 2023-05-18:
Instead of „pulse“ i normally use „Level“ for INPUT 1

closed 2023-06-07 by timoschilling