mongoose-os-apps/shelly-homekit #1281: Calibration fail

C51B, 2023-05-19:
Hi,
With 4 Shelly 2.5 with I have the same issue : calibration always fail and the time movement it set to 5 seconds for a full opening.

[image]

Is it possible to configure the opening and closing time manually?

larsilus, 2023-05-25:
I also thought, that it may had failed, when it stopped at pos 50. i just clicked open, waited for it, clicked save and everything was fine then in homekit.

janHB28201, 2023-06-22:
i have the same problem. Will the problem be fixed?

C51B, 2023-06-22:
No. I give up.

timoschilling, 2023-06-23:
There is a PR with that feature, in that PR is also a beta build linked. https://github.com/mongoose-os-apps/shelly-homekit/pull/449

closed 2023-06-23 by timoschilling