mongoose-os-apps/shelly-homekit #1283: No news?

Rappelsack, 2023-05-29:
I assume that there is no further development here, right?
Nothing new since August 2022. That would be a great pity, of course, because the firmware here is the best available for Shellys. Moreover, there is quite a lot of work behind it, to have come up to here.
Maybe there is a short statement?

GEORGzer, 2023-05-30:
I hope the successors of the Shelly 1 and 2.5 will be supported by a future version 🤗

marmitol, 2023-06-14:
Hi, Thank you very much for the work done. As you comment, I too would like you to continue with the project. For example, now most old devices are not sold like the Shelly 2.5. Just thank you for the good work done. Encourage you not to abandon this project. Thank you very much.

shanghaix, 2023-07-12:
> I hope the successors of the Shelly 2.5, the +2PM will be supported by a future version 🤗

I would recommend using this firmware instead: [https://github.com/RavenSystem/esp-homekit-devices](https://github.com/RavenSystem/esp-homekit-devices)
Needed to replace my shelly 2.5 with shelly +2PM to control my shutters and it works great.

Rappelsack, 2023-07-18:
So can we assume that the Shelly 1, 2.5 will not be further developed here by Rojer?
For a year, nothing more is developed, which is of course a pity, because here that was THE homekit firmware par excellence.

kloibs, 2023-07-27:
Hi everyone! I emailed Rojer personally as soon as I saw this post and I received the following response this morning:

_Sorry for belated reply, but these days, now that I'm officially employed by Shelly, I have stepped away from the homekit community project._

I think it’s safe to say that at least from @rojer‘s side of things there will be no further development here 😞 we can only hope he’s helping the folks at Shelly with their Matter/HomeKit integration!

closed 2023-08-25 by github-actions[bot]