mongoose-os-apps/shelly-homekit #1285: Is the pathway for Shelly HomeKit on Plus devices (ESP32) going to be Matter?

kskenyon, 2023-06-02:
Is the pathway for Shelly HomeKit on Plus devices (ESP32) going to be Matter?

Have a Shelly Wall Dimmer that presently doesn't work with MQTT or HomeKit.

timoschilling, 2023-06-05:
Matter or HomeKit isn't the question. The question is what firmware you want to use. We support Plus devices, the only one we didn't support is the 2PM.

kskenyon, 2023-06-06:
I don't think the Wall Dimmer Plus is supported. Right now I have no way of interacting with it by MQTT or HomeKit.

timoschilling, 2023-06-06:
There didn't exists a Dimmer Plus, I can't find one in the Shelly shop. Did you mean the Dimmer 2?

kskenyon, 2023-06-06:
https://www.amazon.com/dp/B0BLZ79G4Q?psc=1&ref=ppx_yo2ov_dt_b_product_details￼
Shelly Plus Wall Dimmer, WiFi and Bluetooth Enabled with Automatic Calibration, Sleek & Intuitive Touch Slider, Smart Home Automation, Easy to Use, UL Certified
amazon.com

It’s not supported by shelly-homekit as far as I can tell.  Don’t know why.  I read somewhere that it has a coprocessor for dimming.  Maybe that’s the curveball.

I would be happy if the MQTT support from shelly was working on this device but it isn’t.

timoschilling, 2023-06-07:
Sorry my fault, it's a device which is only available in the US market, because it's only 110V.

Maybe someday we support that dimmer, but that will require a developer which has as 110V grid to test it.

Back to your initial question: "Is the pathway for Shelly HomeKit on Plus devices (ESP32) going to be Matter?"
That's a binary question for a non binary problem. I we support the Shelly you have, make a choice depending on your requirements. If we don't support it, take the way which works better for you, use Matter if you can or use a bridge to use a not supported device via homekit.

timoschilling, 2023-06-07:
We will never support Matter. I think you mean the support of the official firmware.

kskenyon, 2023-06-07:
I meant the Shelly Matter release. My mistake.