mongoose-os-apps/shelly-homekit #1289: Shelly 2.5 home kit flashing

wshany, 2023-06-11:
Hello,
I'm a new at this area.First time using Shelly.Upgrade the frimware using the application in my Iphone.
Try to use the script (python) to install the homeKit.failed.

Host: http://shellyswitch25-4C7525060CDE.local
Device Name: None
Model: Shelly25
Firmware: Stock 1.13.0 ➜ HomeKit 2.11.2
Do you wish to flash shellyswitch25-4C7525060CDE to HomeKit firmware version 2.11.2 (y/n) ? y
Now Flashing HomeKit 2.11.2
waiting for shellyswitch25-4C7525060CDE to reboot..............................
still waiting for shellyswitch25-4C7525060CDE to reboot...............
Devices found: 1 Upgradeable: 1 Flashed: 0
can you please help?

timoschilling, 2023-06-11:
Sometimes the state in the python script just stops. What happens if you open the http://shellyswitch25-4c7525060cde.local/ link? Or if you scan for Shelly’s via „./script.py -l“ (l is a low case L)

wshany, 2023-06-11:
This site can’t be reachedshellyswitch25-4c7525060cde.local refused to connect.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

------------------------------------------
C:\Users\william.shany\Downloads\shelly-homekit-master\shelly-homekit-master\tools>python flash-shelly.py -l
Scanning for Shelly devices...

Host: http://shellyswitch25-4C7525060CDE.local
Device Name: None
Model: Shelly25
Firmware: Stock 1.13.0 ➜ HomeKit 2.11.2

Devices found: 1 Upgradeable: 1

C:\Users\william.shany\Downloads\shelly-homekit-master\shelly-homekit-master\tools>

what it mean?

wshany, 2023-06-11:
using the connect with browser"
the firmware shows:
Current version: 20230503-095750/v1.13.0-g9aed950
You have latest version on your device!

timoschilling, 2023-06-12:
Then the flash to HomeKit hasn’t worked. Please try again.

timoschilling, 2023-06-12:
When the script didn’t work, try the flash via url, you find that on the readme. Ensure that the Shelly is allowed to have access to the internet.

xeni2, 2023-07-02:
Hello,
I have another problem with the software. I use the shelly25 at 24V in conjunction with a controller. The Shelly can therefore not measure current and calibration is not possible.

Is there a way to set the time? Currently, this is approx. 5 sec. But need at least 30 sec.
In the HomeKit version, however, the time cannot be set....

Hallo,
Ich habe ein anderes Problem mit der Software. Ich setze den shelly25 bei 24V  in Verbindung mit einer Steuerung ein. Der Shelly kann deshalb keinen Strom messen und das Kalibrieren geht nicht.
Gibt es eine Möglichkeit die Zeit einzustellen? Aktuell sind das ca. 5 Sek. Brauche aber mindestens 30 Sek.
In der HomeKit Version lässt sich die Zeit aber nicht einstellen….

timoschilling, 2023-07-09:
@xeni2 your post is offtopic and will be deleted in the next days to close this issue. But here is a short answer: if I understand it right You have a window cover, that will not work to use it to control a controller. For mor questions, open a new issue

closed 2023-08-08 by github-actions[bot]