mongoose-os-apps/shelly-homekit #1291: Shelly RGBW2 crash and boot loop after changing settings quickly

stosch, 2023-06-16:
So this is my first Shelly and I got it to control a cct led stripe via HomeKit.

Setting it up was easy enough but as soon as I change the values a few times in a short amount of time it crashes and is stuck in a bootloop. Sometimes it recovers itself, sometimes it turns it’s own wifi on and has to be reconnected to my home network. 

As I am writing this it is again stuck in a bootloop since 5 minutes. No way to access it. It just flashes the LED Strip on and off again after a few seconds. It does not matter if I change the values in HomeKit, via Siri or on the web interface. If I am too fast it will just bootloop. If I can help let me know.

stosch, 2023-06-17:
The bootlooping continued for around half an hour. I could not get it to show up anywhere so I cut the power and went to bed. Today I tried it without leds attached but it still bootlooped until it reset itself.

joshuacake, 2023-06-29:
+1 ... I've set initial state to off so device is not rebooting forever and it just crashes the shelly and starts over

adamek333, 2023-07-01:
I have the same issue. :-(

bkleef, 2023-09-22:
Same issue, changed Shelly first to color mode but after flashing it's unreachable by for example `ping`.

bkleef, 2023-09-22:
@stosc @joshuacake @adamek333 have you been able to successfully install an older version and if so which one?

stosch, 2023-09-22:
No, no older version. After hours of rebooting I was able to reach and reset it but that did not help. What helped was setting the initial state to off like @joshuacake suggested. It just crashes and restarts without a bootloop.

kpoineal, 2023-10-19:
I am also having the same issue. Is there an older version of the firmware that doesn’t have this issue? It has made mongoose unusable on my Shelly rgbw2. I reverted to stock and use homebridge but it doesn’t expose RGB+W.

TineTurk, 2024-03-28:
Same issue here. Any plans for an update?

lucassite120, 2024-05-09:
I had this issue too. But I moved to HAA Homekit firmware and my RGBW2 works very stable without any crash. https://github.com/RavenSystem/mgostoHAA

bbrinkerhoff, 2024-05-15:
> I had this issue too. But I moved to HAA Homekit firmware and my RGBW2 works very stable without any crash. https://github.com/RavenSystem/mgostoHAA

Thank you for this. I was unable to have any luck with the OTA install. I believe it may be due to the main '.bin' file being named 'rgbw2.bin' vs 'rgbw2-color.bin'. Changing this and the 'manifest.json' to reflect this didn't get me much farther. 

I ended up just manually flashing the HAA firmware directly over USB-Serial and was able to find success with that method. I've gotten it setup and working and it has been much more responsive and stable so far.

vrpt2022, 2024-05-26:
Good afternoon,
@timoschilling has this been sorted out in the pre-released beta versions?
Thank you

markirb, 2024-05-28:
There is a commit from Timon that should fix crashes at 100% PWM duty cycle. Apart from that I do not know, because I cannot reproduce this effect on my RGBW2

Grufbert, 2024-06-24:
I had the same problem. The new beta has fixed the problem for me https://github.com/mongoose-os-apps/shelly-homekit/tree/2.12.0-beta5