mongoose-os-apps/shelly-homekit #1293: Shelly switches go offline in home app intermittently but often

knanyahu, 2023-06-19:
I have a rather large HomeKit setup with a mix of iDevices and Shelly switches running mongoose firmware.   Im running Aruba Instant On wireless APs and switches.  The switches intermittently go offline in the apple home app but it happens quite often.  It seems the Shelly switches are much more affected than the iDevices switches.  I can see the switches on the network (ping able) and i can browse to the switch.  But they are unresponsive in apple home.  If i reboot the APs, it often fixes the issues.  What do you think?  How can I improve the reliability of these switches on HomeKit?

bobf41, 2023-06-28:
I am having the same kind of issues with a 1pm but a Unify is running stable. 
The 1pm just disappears from the WiFi network. 
Put a simple AP at 2.4Ghz yesterday and reset the 1pm since it didn't came back after turning it off and on. 
I connected to the AP and tried some reboots of the AP as well but after an hours the 1PM was disconnected again. 

The unify is very stable at another location but in the same type of WiFi network.

confuu, 2023-07-10:
I'm having weird problems with Shelly 1L switches too. I have them in detached mode and push button automations stop working time to time. When they stop working in homekit you can access them through web browser normally and they also answer to ping of course. Wifi signal is good -30..-50 and there does not seem to be really any other interference on channel 11 (2.4ghz). AP reboot fixes the problem for me too.

bobf41, 2023-07-10:
> I am having the same kind of issues with a 1pm but a Unify is running stable.
> The 1pm just disappears from the WiFi network.
> Put a simple AP at 2.4Ghz yesterday and reset the 1pm since it didn't [...]

I found the issue. The power was 22-22,5v. I assume the Shelly plus 1 isn't running stable on this. Switched to 230v and issue is gone. Shelly 1 plus is set to run from 24v. I wasn't getting this from the gate controller.

closed 2023-08-09 by github-actions[bot]