mongoose-os-apps/shelly-homekit #1297: /rpc/Shelly.GetInfiExt

DavideTommasin, 2023-07-01:
Can I have an example how to call the api Shelly.GetInfoExt with auth_eh = true in post or get mode? How to pass the password? Thanks

timoschilling, 2023-07-01:
That depends on how you want to call the api.

DavideTommasin, 2023-07-02:
I try to call via http with rest call, now i can try the call with postman (but all times i have 401 unauthorized). When i understand how to pass the password i would call in swift code, but step by step first of all is postman 😁

DavideTommasin, 2023-07-04:
call a method via http (for example /rpc/Shelly.GetInfoExt) this call return an error (empty body) but in the header return WWW-Authentication, inside there are nonce, use sha256 for calculate the digest auth, then call via http (/rpc) with this body
{
    "id": 1,
    "method": "Shelly.GetInfoExt",
    "auth": {
        "realm": "shelly1-414647",
        "nonce": 1688452777,
        "username": "admin",
        "cnonce": "80217314",
        "algorithm": "SHA-256",
        "response": "45b2a3b12ff0ee47221f9fc1118567f3d4fb7246d715ef033450c55a852f6ea4"
    }
}
and authentication work right

closed 2023-07-04 by DavideTommasin

timoschilling, 2023-07-04:
Thanks for documenting and maybe helping others.