mongoose-os-apps/shelly-homekit #1298: HomeKit widget not reflecting on/off (switch state) for Shelby 1

i-am-meikle, 2023-07-03:
Hi, 

I have a Shelby 1 (v3) which is connected to a switch and a light. I have tried it both with and without the switch connected. 
I flashed the HomeKit fireware (which worked very painlessly 👍) and included the Shelly 1 as a Light in HomeKit. 
This all worked fine and I can see the Light in HomeKit. 

I can operate the Light from the Fireware homepage (The Shelly 1's own webserver) OK and the status of the switch is correctly reflected on the website if I manually operate the switch. 

In the HomeKit app, I can see the Shelly 1 as both a light and a switch

[image]

If I select the "Light" it turns on/off the light as expected.

If I select the Switch

[image]

When I try and toggle the switch, it switches the light on/off but always immediately returns to show the light as "off".
Is this a config problem on my side, or an issue ?

Thanks
Ian

i-am-meikle, 2023-07-03:
Humm ... strange.
Literally just after I posted this, HomeKit crashed on my phone and when I restarted it, the Shelly 1 is now shown as a Light (not a switch). I did no changes at all.

[image]

It now works as expected.
Anyone else experience this ?

jactmp, 2023-07-03:
Not this. But since introduction (degradation?) of new Home architecture in iOS 16 I experience some other issues. Quite often have to restart Home app (e.g. after pairing new device) to actually see the changes. Now I'm doing it every time - just in case (e.g. to avoid such problems like yours).

timoschilling, 2023-07-03:
Solved

closed 2023-07-03 by timoschilling