mongoose-os-apps/shelly-homekit #1299: Shelly 1 - too many Home app notifications!

stevenperich, 2023-07-05:
Hey there

I have a Shelly 1 that's now running 2.11.0. I've had it running for > 1 year and it's been glitch free except for ONE thing that I haven't been able to crack - though it might be something on the Home app side rather than Shelly-Homekit.  It's running as a garage door opener and I get too many iOS/Mac notifications.  Instead of just getting a notification when the door opens (and the Shelly detects the reed switch has moved), then another when the door closes, I get one notification saying the garage door opens. Then, when it's shut, I normally get 3 notifications - one saying it's closed, another saying its open, and another saying its closed!  Any way to remedy this from your side, or do I need to go hunting a fix on the Home app side?

Thanks in advance (and I am about to donate to your project - I meant to ages ago and forgot. you have saved me $100s not having to install the door opener manufacturer overpriced option!)

timoschilling, 2023-07-09:
Double of #1087

closed 2023-07-09 by timoschilling

stevenperich, 2023-07-10:
Omg thank you Timo! I’ve been trying to work this out for ages. Will change
the timers and see if that fixes it. Thanks!