mongoose-os-apps/shelly-homekit #1304: Devices disapearing from Home

i-am-meikle, 2023-07-30:
I have 4 Shelly devices on my network. 
2 Shelly Plus 1
1 Shelly Socket
1 Shelly H&T

All are flashed with the Mongoose firmware. 
Twice now, after almost 2 weeks or working flawlessly, all devices apart from the H&T disappear from Home. They do not go "offline", they just disappear. The HomeKit settings say they are Paired, but there are no Active connections.

I have to reset the connection and start again. 
This is not good. 

Is anyone else experiencing this ? What can the cause be ?

I have several Eve devices on the network too, and they work faultlessly. 

Thanks
Ian
(I would add a picture of the web GUI, but I cannot work out how here ?)

timoschilling, 2023-08-01:
Yes the H&T is not supported by us.

@i-am-meikle pleas post a screenshot. Just click hier to upload.
[image]

i-am-meikle, 2023-08-01:
Regarding the H&T, correct, ,my mistake, I have this through HA hence probably why it did not drop with the others. 

The socket is a Shelly Plug, **NOT** the Plug S

(I cannot believe I did not see the image attach option ...)

[image]

i-am-meikle, 2023-08-07:
I am wondering if this has anything to do with auto-update function in Home. Every time this problem has happened to me, I have had a message in Home to turn on AutoUpdate for accessories.

i-am-meikle, 2023-08-29:
Am I the only one experiencing this ?

closed 2023-09-27 by github-actions[bot]