mongoose-os-apps/shelly-homekit #1308: Shelly 2.5 disconnection after a few days form WIFI

samuelroeben, 2023-08-09:
Hi, i have a hard time with my 2.5 shellys. 
They are disconnecting after a few days or even some hours. If this happens i have do connect manually to them with 192.168.33.1 and enable Wi-Fi again.   They are all setup with the new Firmware.
The Wifi is always on and has WPA 2 + 3 and if i enable them again they are working absolutely fine. 
Power saving is disabled. 

I also have some shelly 1 in use and they don´t have this issue. My first thought was a hardware defect or something, but I think this is unlikely cause all of them have this issue. 

Maybe there is something else i can do to find the error or misconfiguration.

d4rkmen, 2023-08-10:
Hi, reporting the same happened to me with shelly 1
Problem solved by moving from 24V to 220V AC power source

stanislavsulima, 2023-08-11:
I have the same issue with Shelly 2.5. Multiple Shelly 1 work stable.

jotzet79, 2023-08-31:
Same here with Shelly2.5 / Shelly1PM, random WiFi disconnects. After restarting either Shelly or Router up again…

anybody knows if reverting to stock resolves those issues?

samuelroeben, 2023-09-01:
I have found kind a way around. I reflashed the shelly 2.5  and setup the same wifi also as backup connection. This works reliable since 2 weeks. Another 2.5 has some kind of hardware problem. Cause i can´t setup the wifi with the orginal shelly software for flashing to the homekit version. I can connect to it, but 2 seconds later the accespoint is off.  I also bought a serial to usb converter for flashing via cable. With this i have a problem with the driver to get it running with my mac. The Stock firmware doesn´t have solved the problem for me. That is why i think it is a hardware issue in my case. The shelly is also kind a noisy.

stanislavsulima, 2023-09-07:
In my case, after restarting unresponsive Shelly, the light goes off/on by itself after 5-10 mins. It may be, indeed, a hardware issue. Speaking of WiFi, I am still looking for a solution. The worst thing is that buying a new `Shelly 2.5` is almost impossible, while `Shelly 2PM` and `Plus 2PM` are unsupported here.

closed 2023-10-19 by github-actions[bot]

jotzet79, 2023-11-10:
Honestly we cannot let this being closed, as it is still a serious issue with some of the the devices!
However I encountered 2 different oddities:

- Some shells have bad capacitors in them and after some time / after power failures in your home they simply die: I recorded such a dying poor thing -> https://www.youtube.com/shorts/fRvjT94Bq7g
- Some shellys are disconnecting from wifi after some time (x hrs to x days); I suspect that this has to do with wrong Wifi settings / bad Wifi Router configs (in my case Asus RT-AX58U).

DIRTY! Today I have setup cron jobs on my raspberry PI 4B (running Raspberry OS 64bit Debian 12 Bookworm) to restart the affected shellys (shelly1pm & shellyswitch25). Here every 20 minutes the shelly is triggered to reboot via rpc call using cURL. Let's see if this "fixes" things. Personally / professionally I absolutely hate to restart devices & services, but what can I do now? 

I'm totally frustrated already having to restart shellys or reboot the Wifi router in order to bring Homekit Scenes to live again - every few days.

timoschilling, 2023-11-13:
For those who think it's a wifi issue in the Shelly please file a issue on https://github.com/mongoose-os-libs/wifi

jotzet79, 2023-11-14:
> For those who think it's a wifi issue in the Shelly please file a issue on https://github.com/mongoose-os-libs/wifi

Sure, but isn’t this project also rather abandoned? Seems that @rojer is not active anymore since he joined alterco. 

I mean I totally understand if sb does not have the time (anymore) to spend his/her free time on open source projects and activities anymore, given the stressful times we are facing in our daytoday business struggles and private lives…

samuelroeben, 2023-11-24:
I found an alternative. I will ditch most of the shellys that make me trouble. There are new meross devices with the same functionality and native homekit support.

x23piracy, 2024-01-03:
Hi,

i you may have 2.5 with Chongx capacitors the 100/16 is false and need to be replaced, there are also 2 high voltage capacitors (400V) which are also broken in 1 of cases, i have 3 2.5 and all of them were rustling, sounds like flowing water in the wall, i replaced the 100/16 capacitor at all of them and they all went back to life.

Because of the broken capacitor the coil starts to make noise and the mains adapter get confused which stops the shelly from working.

[image]

[image]

https://www.shelly-support.eu/forum/thread/14612-defekte-elkos-bei-shelly-2-5/

Best Regards X23