mongoose-os-apps/shelly-homekit #1313: Support for Shelly Mini?

perstromqvist, 2023-09-15:
Will the new Shelly Mini's be (listed as) supported by Mongoose?

rionshin, 2023-09-17:
@timoschilling  Does the new mini devices use same chip, do you plan to support y ? 

https://www.shelly.com/en-bg/products/shop/shelly-plus-1-mini

timoschilling, 2023-09-21:
I don’t have a test device, but possibly it will. The mini 1 would be a problem I think. The mini EM and mini 1pm would be harder.

petereit, 2023-11-09:
In the mean time, I added my 1 Plus Mini's to HomeKit by installing Homebridge and Mosquitto MQTT broker on an old Mac Mini that I use as an AirPrint server.

Larsn1, 2023-11-09:
Well I think that’s a good solution bc the plus minis get matter if Shelly releases a firmware update for it

Larsn1, 2023-12-01:
Updated mine to Tasmota 13.2 which supports matter. Is the method of injecting the bootloader really so different?

juliennnnn, 2023-12-03:
Hello,

Do you plan to support the Shelly Plus 1 Mini ?

Thanks a lot

lucassite120, 2023-12-19:
Finally, I have my Shelly Mini working in Homekit with HAA firm: https://github.com/RavenSystem/esp-homekit-devices

I followed these steps: https://www.youtube.com/watch?v=tXkaTqY86cc

timoschilling, 2023-12-20:
> I'd be more than happy to sponsor both a Shelly Plus 1 Mini as well as a Shelly Plus 1PM Mini to you :)

@cpm1 write me a email, address can be found on my profile.

cpm1, 2023-12-28:
> @cpm1 write me a email, address can be found on my profile.

FYI, did so on 21st of December. If you didn't get it, please check your spam drawer.

Schw3n, 2023-12-28:
I also sent a mail in 21st Dec. to offer also a sponsoring.

Schw3n, 2023-12-28:
> > write me a email, address can be found on my profile.
> 
> @timoschilling FYI, did so on 21st of December. If you didn't get it, please check your spam drawer.

I also sent a mail on 21st Dec. to offer also a sponsoring.

skyedrummer, 2024-02-13:
Any news on the Mini support? Currently using HB for them, but this is best.

SchroedersKater, 2024-02-22:
If it helps, I’d like to sponsor a Shelly 1PM Mini Gen3 as well. Just drop me a PM. BTW, where are you located? To some countries shipping costs are more than the device costs.

DHEVAN1, 2024-03-17:
Hi, Just wanted to know if there is any update on support for Minis, and if you think its possible to happen.

g3ntleman, 2024-03-22:
I accidentally ordered a Mini, gen3. Specs say it has a "ESP-Shelly-C38F" CPU.

lukasfoempe, 2024-03-25:
Hi all,
is there any update for supporting the 1mini gen3?
Would be great.

br
lukas

lucassite120, 2024-04-11:
HAA Homekit firmware currently supports Shelly Mini, even gen3. I flashed 2 Shelly Mini 1 gen3 using OTA and they works like a charm. You have a video with subs explaining all process:
https://www.youtube.com/watch?v=06YHkRkwJE4

I3lackJ4c, 2024-06-05:
@timoschilling is there any new on adding the new Shelly 1 Mini Gen 3 to the list of Homekit support?

DHEVAN1, 2024-06-05:
> @timoschilling is there any news on adding the new Shelly 1 Mini Gen 3 to the list of Homekit support?

I have an idea and I haven't bought a mini gen 3 yet but if you would like you can try it. You can. You can add it to home assistant and have home assistant running. Then you can get the HomeKit plugin on home assistant which should make your home assistant devices work on HomeKit.

tobz-nz, 2024-06-05:
> I have an idea and I haven't bought a mini gen 3 yet but if you would like you can try it. You can. You can add it to home assistant and have home assistant running. Then you can get the HomeKit plugin on home assistant which should make your home assistant devices work on HomeKit.

I've done this, and its working great. I've got 3 mini gen 3's and a H&T sensor all in HomeKit via HA

I3lackJ4ck, 2024-06-05:
Hello,
thanks for the answer. 
I want to avoid having to setup a Homeassistant or homebridge with tasmota or anything like that as i have no need for more usage other than a switch. I already have 10 shellys with the mongoose implementation and they all work great. Seems like i have to wait or try to get one of the older Shelly 1s

DHEVAN1, 2024-06-06:
> I've done this, and its working great. I've got 3 mini gen 3's and a H&T sensor all in HomeKit via HA

Can you send a picture of what the HomeKit interface looks like with the Shelly's on there. I am planning on buying 10+ Shelly mini gen 3's

tobz-nz, 2024-06-06:
@DHEVAN1 This is one room. The `Lights` is a Shelly mini & the Temp & Humidity are the H&T sensors.
The heater and lamp are other things.

[image]

metaljay, 2024-06-24:
Is this even being worked on anymore? I only ask as I can’t see a development branch or any comments regarding support for the Gen3 products.

metaljay, 2024-06-25:
> Hi,
> 
> I do not think someone (@timoschilling ?) actively pursues Mini Support at this time. Though it should most likely not be that hard to add support, this takes time and resources which are limited for us currently .)
> 
> If you want to support, then you can do so with (in order): Writing Code, Donating Hardware, Testing, Money....
> 
> We will release a new version soon (Support for +2PM), and maybe Shelly Gen3 is the next step once this is completed.

I can help with two of those things - money and donating hardware.

I’ve already emailed @timoschilling offering money as I personally have used his firmware many times. 
If there is anyone else actively giving this ago (and willing to do a pull request) I’ll happily donate to you instead

markirb, 2024-06-27:
Hi.

Let's talk after the current release, I hope this weekend.

Maybe Timon will give it a go, maybe I can do it. Let's see

lucassite120, 2024-06-29:
I used HAA HomeKit firmware with my Shelly Minis. It's very easy to install using OTA and works great. Check it out:
https://github.com/RavenSystem/mgos32toHAA

iiseppi, 2024-07-10:
When testing to support Shelly Mini's start, does that mean also that, new GEN3 devices is getting support?

markirb, 2024-07-10:
It would mean that the way to support other devices is not far.

@metaljay I sent you an email.

markirb, 2024-07-16:
Looking good.... first version is running. 

Things not working so far:
Temperature NTC (library not yet compatible with esp32c3)
PWM (library needs to be ported; not needed actually)

Still some work to do, especially to incorporate the workarounds into the normal build environment.
If anyone wants to try (use at your own risk). Revert did work for me to and from 1.3.3:

[fw.zip](https://github.com/user-attachments/files/16254817/fw.zip)

metaljay, 2024-07-16:
> Looking good.... first version is running.
>
> Things not working so far: Temperature NTC (library not yet compatible with esp32c3) PWM (library needs to be ported; not needed actually)
>
> Still some [...]

Posts like these bring me so much happiness (sad I know)
You only got the minis the other day and you’ve already got a working version! Well done!!

markirb, 2024-07-16:
I mean, it was really not hard to do. It just worked...

Which is surprising since it is RISC-V now. But rojers lasts commits for mongoose-os that has since been kind of abandoned, did most of the work...

markirb, 2024-07-17:
This app never has and almost certainly never will support Shelly App/Cloud

DHEVAN1, 2024-07-17:
Alright thanks!

lucassite120, 2024-07-18:
Since [HAA firmware](https://github.com/RavenSystem/mgos32toHAA) adds HomeKit suport to all new Shellys with infinite configurations, having Shelly App/Cloud with Mongoose would be a great feature.

lucassite120, 2024-07-18:
> Yeh that’s great, but not something they are looking at (and I don’t blame them for the niche population that want it). I’d much rather have this firmware than the HAA one. If you’ve used this firmware on Shelly 1s before vs the HAA one you’ll know how lightweight this is and how the response times are incredible.

I have used both firmwares, and HAA is more light and has better response time by far.

metaljay, 2024-07-18:
> I have used both firmwares, and HAA is more light and has better response time by far.

Then don’t clog up a discussion promoting another firmware (that ultimately requires a paid app on iPhone / android to configure easily).

markirb, 2024-07-18:
unfortunately all that's @lucassite120 is doing: spamming several topics promoting HAA.... @lucassite120 I do not know what your interest is, but I highly doubt your motive is to help users so please stop it

lucassite120, 2024-07-18:
I am not promoting HAA. I was just trying to help users because this firmware is practically dead since Rojer left. I have tried the new version and it is even slower than the latest Rojer's version. I have used this firmware since the beginning, but it is a pity what it is now. I only asked about Shelly's cloud feature because it is something that HAA will never have. I have over 120 Shelly devices and using the Shelly app along with HomeKit would be a great point. Don't worry, I'll leave this community of fanatics. The HAA community is much friendlier. FYI: HAA App is not needed, I only use it to batch updates. ASAP I will move to HAA the rest of my Shellys. Bye!

markirb, 2024-07-18:
You may have the feeling that it is slower. Do you have numbers? I doubt it to be slower since nothing changed underneath. 

Mongoose is still the same. And even basically the same that Shelly itself uses until now based on IDF 4.4. Shelly seem to continue their very own mos fork now. But only in 1.4.0 (currently in beta) they migrate to newer IDF.

Users can choose whatever they want, there are several firmware alternatives and each have their up- and downsides. If you migrate for now liking this community then fine. 
I only feel that it is not very helpful if you post under every
Thread what you use. 

If you want this firmware to be faster and better then any help is appreciated.

danielowenuk, 2024-07-24:
@markirb I loaded your firmware to a mini1, appears to be working perfectly

I did have a few linked to homebridge I always prefer native homekit where possible.

VERY much appreciated.

skyedrummer, 2024-07-24:
Does this firmware work on Gen3 devices?

metaljay, 2024-07-24:
> Does this firmware work on Gen3 devices?

There’s a beta firmware that you can compile yourself that supports it, or wait a few week/s and it will be published as general release (there’s an active pull request ongoing now)

skyedrummer, 2024-07-30:
Try as I might I can't get the fw.zip to load into the shelly mini from its GUI. Mini is on 1.3.3. If I update shelly to 1.4.0 beta, it flashes that it is uploading to the mini and then nothing. Any ideas?

markirb, 2024-07-30:
Update from 1.4.0 is currently not possible. I have a campatible brach, but that's nothing for now. Go back to 1.3.0.
fw.zip from above is only compatible with mini1 gen3

skyedrummer, 2024-07-30:
Ah, okay, I think the mini I’m trying to update is not a gen3 device. Will the finished firmware be compatible with all the versions?

markirb, 2024-07-30:
Was not planned. But will be very easy. Cannot promise but maybe after my summer vacation

markirb, 2024-08-29:
Support for Minis is there in latest beta available in releases section: Mini 1 Gen3, Mini 1PM Gen3 and also Plus 1 Mini.
Please test

pfrasun, 2024-08-29:
Just tested 2.12.2_beta1 with Shelly Plus 4 - after update it misses INPUTS section in UI.

markirb, 2024-08-30:
> Just tested 2.12.2_beta1 with Shelly Plus 4 - after update it misses INPUTS in UI.

Thanks you're right, totally unrelated javascript error. Is fixed in beta2 which is up now

markirb, 2024-08-30:
Which file do you have? You have to update from 1.3.3. from 1.4 it is not possible atm

skyedrummer, 2024-08-30:
I downloaded this file  - shelly-homekit-ShellyPlus1Mini.zip  
from
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.2-beta1.

I'm on 1.3.3 - should I upgrade to 1.4.2? or do you mean go to 1.3.3 from 1.4?

markirb, 2024-08-31:
> edit - ok, I got mixed up with your response, I can only update from 1.3.3 have i got the correct firmware file though?

Yes this should be the correct firmware. You upload this zip in the firmware section of the shelly. If this does not work then please enable debug over WebSocket and Watch Logs in Diagnostics.

skyedrummer, 2024-08-31:
here is the log output (if it works)

[diagnostics-shelly-plus1mini-debug-log.txt](https://github.com/user-attachments/files/16825871/diagnostics-shelly-plus1mini-debug-log.txt)

markirb, 2024-09-01:
> here is the log output (if it works)
> 
> 
> 
> [diagnostics-shelly-plus1mini-debug-log.txt](https://github.com/user-attachments/files/16825871/diagnostics-shelly-plus1mini-debug-log.txt)
> 
> 

Thanks. That does help indeed. Will need to build a new version

markirb, 2024-09-02:
This here should work: 
[shelly-homekit-ShellyPlus1Mini.zip](https://github.com/user-attachments/files/16836725/shelly-homekit-ShellyPlus1Mini.zip)

skyedrummer, 2024-09-02:
Thats seems to have done the trick, brilliant, thanks for all your work!