mongoose-os-apps/shelly-homekit #1317: Console error when clicking the "Setup" button on a Shelly i3

bkleef, 2023-09-21:
1. Just update from the latest Shelly stock firmware to: http://x.x.x.x/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyI3.zip;
2. Disabled input 2 and 3;
3. Clicked on the _Setup_ button (from Chrome) and got:
> Uncaught TypeError: Cannot read properties of undefined (reading 'toLowerCase') at el.onclick ((index):1424:27)

Which is the last line of:

```js
el("hap_setup_btn").onclick = function() {
  el("hap_setup_spinner").className = "spin";
  // Generate a code from device ID, wifi network name and password.
  // This way it remains stable but cannot be easily guessed from device ID
  // alone.
  let input = lastInfo.device_id + (lastInfo.wifi_ssid || "") +
      (lastInfo.wifi_pass_h || "");
  let seed = sha256(input).toLowerCase();
```

Please help. Happy to test something!

bkleef, 2023-09-21:
Exactly the same error with 2.10.1 but on a different line.

> Uncaught TypeError: Cannot read properties of undefined (reading 'toLowerCase') at el.onclick ((index):1132:27)

bkleef, 2023-09-21:
These are the relevant values used in variable `input` received over RPC. Maybe it has to do with the less common characters in the SSID?

```json
{
    "id": 8583,
    "src": "shellyix3-E8DB84D29388",
    "result": {
        "device_id": "shellyix3-EADB84D29388",
        "wifi_ssid": "ευχαριστια (2,4 GHz)",
        "wifi_pass_h": "531c54f836448b73c580bbacae32bd23"
    }
}
```

bkleef, 2023-09-21:
The non ASCII characters in SSID are definitely the cause of this bug as the sha256 function uses https://github.com/geraintluff/sha256 which mentions:

> Input must be an ASCII string - if character codes outside the range 0-255 are received, undefined is returned.

What about:

1. Instead of `wifi_ssid` use a key like `mac_address` or `wifi_conn_ip` as we know for sure that these keys will only contain ASCII values;
2. Test if the SSID is a valid ASCII string before using it in `input`:
```js
function isASCII(str) {
    return /^[\x00-\x7F]*$/.test(str);
}
```
3. Remove `wifi_ssid` from `input` variable.

timoschilling, 2023-10-16:
We already hat that issue multiple times, so we should take a deeper look on that. Not sure on the impact on existing setups and about maybe needed changes in the C code.

bkleef, 2023-10-16:
If you use option 2, nothing would change in existing setups. I tested option 3 on ShellyI3, Shelly25 & ShellyPlugS and works fine.