mongoose-os-apps/shelly-homekit #1319: Plus1PM on stock 1.0.3 is unable to be flashed

dannycpw, 2023-09-27:
Hi Team,

I am unable to flash the firmware on my +1PM & I guess that might be the stock version is too new, I guess. Below is what I've done.

1. script is unable to discover my Plus1PM my my Mac

Scanning for Shelly devices...
Devices found: 0 Upgradeable: 0 Flashed: 0

2. By putting http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1PM.zip, +1pm stated is updating and reboot afterward. However, it still boot up with the stock.
3. Same as above, if I put "http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1PM.zip" into "Update from a web server" option.

Thanks & regards,
Danny

timoschilling, 2023-09-29:
I can only say something to point 1. at the moment: The flash script didn't support the Plus devices at the moment.

dagaloni, 2023-10-17:
I had the same problem. I tried to restore the the firmware using this page: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#reverting-to-stock-firmware

But that didn't work. I got an error saying the update failed. After that I used this link and that worked! http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update

Hopefully this will work for you too (maybe try updating the firmware even though it will fail).

dannycpw, 2023-10-18:
I finally able to flash it & now on HomeKit natively!

I wonder the url on the README.md is not working, but those on the Release page (https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.11.2)!

Thus, I use below URL to flash my +1PM
http://A.B.C.D/ota?url=https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.11.2/shelly-homekit-ShellyPlus1PM.zip

Another interesting thing is that I have 6 x +1PM, 5 of those can be flash with a single trial. But one of those is fail with the same symptom as before! I just keep trying for 6 or 7 times and it can be flash suddenly! Don't ask me why as there's no log can be captured as well as all those 6 +1PM are on the same stock firmware & bought on a single batch.

Hope this help!

timoschilling, 2023-10-19:
Interesting to know that `https` urls like the GitHub release url works on plus devices. Gen 1 devices can't flash from there.
That it sometimes needs many tries is a common issue.

closed 2023-10-19 by timoschilling