mongoose-os-apps/shelly-homekit #1321: Support for Shelly Plus 1 Mini

alessiobulfaro, 2023-09-29:
Hi,
Shelly newly released mini versions of some Shelly Plus devices, such as the Shelly Plus 1 Mini.

Basically it is the same as the Shelly Plus 1 (https://shelly-api-docs.shelly.cloud/gen2/Devices/ShellyPlus1/), but has an own device model ID.

I've two test device if needed. Hope support for these devices will be added easily.

geekpav, 2023-10-16:
Stay away from the MINI version of the Shelly Plus 1 PM if you are using Homekit... There is no way to flash the module to a Homekit firmware. One thing you can do is to use it through a Homebridge server, however it doesn't work very nicely : it appears doubled on the "Apple" Home interface. I replaced my Shelly Plus 1 PM MINI modules with ordinary Shelly Plus 1 PM, flashed to Homekit, which work seamlessly.

timoschilling, 2023-11-02:
double of https://github.com/mongoose-os-apps/shelly-homekit/issues/1313

closed 2023-11-02 by timoschilling

djcargo, 2024-01-24:
I'm using some Shellys Mini with Home Assistant, which puts it to HomeKit and it works fine.

DHEVAN1, 2024-06-01:
How do you make it to where it can be used in HomeKit when it's running on home assistant?

djcargo, 2024-06-01:
I have used link from here: https://www.home-assistant.io/integrations/homekit/

There's button "Add integration to...", and this integration showed in my Settings->Devices&Services