mongoose-os-apps/shelly-homekit #1322: Reverting and Controlling issues

Imburdalinix, 2023-10-01:
So there are multiple issues I've been facing recently involving the Shelly 2.5 with HomeKit.

For starters, if the roller shutter is at about 20% you can easily control it by taping on (for example) 80%. But as soon as you tap on 100% it doesn't go up, but down. It reverts the input on a full value like 100 or 0... I made a video trying to demonstrate the problem. Click [here](https://www.icloud.com/iclouddrive/048juAdcv-0wAcKte0Ajngx0A#Shelly_Bug) or on the link below to view the video.

https://www.icloud.com/iclouddrive/048juAdcv-0wAcKte0Ajngx0A#Shelly_Bug

Imburdalinix, 2023-10-09:
I would be happy to get a response ☺️

Teimotronic, 2023-11-02:
Same problem here. Even if I slide the slideer, the roller shuttle stops at random positions.

ramosmfilipe, 2023-11-14:
I have the same issue, slider just jumps to random positions. only "open all" and "close all" automations via homekit work properly. Please help.

Imburdalinix, 2023-12-06:
Hello, is it possible that anybody could respond to this problem? It has been almost 3 months since this post has been made. If you could just explain how you're currently progressing instead of staying silent that would be great.

Teimotronic, 2023-12-06:
The problem still remains

Imburdalinix, 2023-12-19:
Okay, so I've seen that Timo Schilling has seen the video. Could you, mr. Schilling, please respond to this issue or at least give us an update? 
Thanks!

lucassite120, 2023-12-22:
I had issues with my roller shutters too. Since Rojer leaves development of this firmware, I researched alternatives and I found this firmware: https://github.com/RavenSystem/esp-homekit-devices/wiki

It's actively developed and my shutters work like a charm. I hope this can help you.

closed 2024-01-21 by github-actions[bot]