mongoose-os-apps/shelly-homekit #1323: RGBW2 / Control with Switch not possible

BA0815, 2023-10-05:
Hello team,
I have a problem with the button controls after flashing my shelly rgbw2.
The control works with the original software (button type edge switch).
After flashing on Shelly Homekit, the control via the wall switch does not work. The LED strip then just starts to flicker.
Switching on/off no longer works.

Is there anything to consider when flashing?
What am I doing wrong?

Electricity Europe 230V AC

timoschilling, 2023-10-07:
Are you using the AC/DC converter (black with two wires on each side)?

BA0815, 2023-10-07:
yes, I used the included converter.

timoschilling, 2023-10-08:
We do not support that. Change your setting to not use that adapter or use the stock firmware

closed 2023-10-08 by timoschilling

przemekpobuta, 2023-10-09:
@timoschilling Do you plan to support that in the future?

timoschilling, 2023-10-09:
We do not have a explicit plan to support it. But that not mean at never will happen. Maybe someday someone will implement it.