mongoose-os-apps/shelly-homekit #1325: Shelly 1L Responive in WebUI, unresponsive in Homekit

ryzuryzu, 2023-10-12:
Hello, 

I just added my first shelly to bedroom light switch.
Been able to accès it via webUI after flashing HomeKit software. Worked as a charm.
After adding it to HomeKit it instantly showed "no response". (added as Input 2 gate + shelly sw switch)

What I did:
Delated and added it to HomeKit 5 times, nothing changed
reset power in that switch, nothing changed
reset router - now after resetting it has another ip address. With that new dress i'm unable to access it via webUI too :(

Any help?

timoschilling, 2023-10-12:
I can’t see the screenshots. They didn’t load. You can simply drag and drop them into the comment field or use the image add button above the comment field.

ryzuryzu, 2023-10-13:
There are links to image hosting site in ma post :)

timoschilling, 2023-10-13:
Yes but the screenshots didn’t load on the page, it only shows advertisements and request a payment if I try to download them. I remove the link to prevent from a possible cam and close this issue until you provide accessible screenshots.

closed 2023-10-13 by timoschilling