mongoose-os-apps/shelly-homekit #1328: shelly 1plus whit add-on signal out issues

aljoh20, 2023-10-29:
hello 

"I'm attempting to implement a solution using a Shelly device to activate a diesel heater when the ambient temperature falls to 2 degrees. I've already connected a Shelly Add-On temperature sensor to my Shelly 1 plus device, which can now monitor the temperature.

My question is, how can I configure the Shelly device to trigger a specific protocol code (11) and send it through the TXD and RXD connections to turn on the diesel heater when the temperature reaches 2 degrees? I also have a Shelly Uni Chip available, which may serve as an intermediary in this communication.

Could someone provide guidance on how to correctly configure the Shelly device, the Shelly Uni Chip, and the communication with the diesel heater to achieve this?"

rionshin, 2023-10-30:
Mongoose Expose the Shelly key as standard relay to apple home , how you will use it - it's your decision. The Shelly cannot send codes to TXD/RXD connections as it is simple replay - on/off.   You can send 1/0 (On./off) to UNI and then uni to send what ever is needed to your diesel heater.  The automation can be created in Apple home.   You should search for help in apple forums related to Automations and Home shortcuts.

timoschilling, 2023-11-06:
We do not support any TX/RX communication.

closed 2023-11-06 by timoschilling