mongoose-os-apps/shelly-homekit #1333: Shelly Plus1 not connecting after reverting to stock fw

BeigeRonin, 2023-11-08:
Hi. I needed to revert one of my Shellys back to the stock firmware, but in doing so, the Shelly is now just stuck in an unaccessible state. The screen just displays "Connecting to Shelly" but nothing happens. It's still connected to my Wi-Fi network.

Any ideas please?

timoschilling, 2023-11-09:
Where do you exactly see the "Connecting to Shelly" text? Please post a screenshot.

BeigeRonin, 2023-11-10:
The browser when I try to connect to the Shelly...

[image]

BeigeRonin, 2023-11-13:
> Where do you exactly see the "Connecting to Shelly" text? Please post a screenshot.

Hi Timo - see above. Do you have any idea what I can do with the Shelly to recover it, or should I just replace it and start again? Thanks

timoschilling, 2023-11-13:
Sorry I have no idea.

Ju17091, 2023-12-05:
Remove the fuse briefly. And then switch the shelly several times using the light switch until you hear a click. Shelly then returns to the factory settings and can be set up again via WifiAP

closed 2024-01-04 by github-actions[bot]