mongoose-os-apps/shelly-homekit #1334: When Garage state is "stopped", Homekit shows "Closing" indefinately

DanielNagy, 2023-11-09:
Shelly Plus 1, with single sensor for Closed status.

To replicate:
When Garage status is closed, press "Toggle" from the WebGUI to open door, then press "Toggle" again after a few seconds (but before the Movement timeout value is reached).
This can also be replicated from the Apple Home App.

Garage Status will now show "stopped", which is correct, door will be partially open, but the HomeKit status will show  "Closing".

[image]

Furthermore, the Apple Home App throws even more confusion into the mix.
The Garage Badge shows it as "Open", with a small "Closing" status under security.
[image]
However the detail view adds confusion as it shows the selector switch in the Closed position, with a status above showing "Closing..."
[image]

The Garage Status of "Stopped" is correct. as the motor stops in in a partially open position. So Homekit should show "Open" (without the "Closing" status).

```
8091391221 shelly_main.cpp:449     Up 8091.38, HAP 0/1/16 ns 1, RAM: 211588/196900; st 37; 5.1: cur:closed tgt:closed cl:1 op:-1
8097307201 mg_rpc.c:314            Shelly.SetState via WS_in 192.168.1.86:57995
8097312626 shelly_output.cpp:64    Output 1: off -> on (GDO:open)
8097319225 shelly_hap_garage_d:309 GDO 1: Tgt State: closed -> open (1 -> 0) (RPC)
8097325587 shelly_hap_garage_d:224 GDO 1: Cur State: closed -> opening (1 -> 2)
8097633580 shelly_output.cpp:64    Output 1: on -> off (pulse_off)
8097923378 shelly_input.cpp:77     Input 1: change (state 0)
8099391625 shelly_main.cpp:449     Up 8099.38, HAP 0/1/16 ns 1, RAM: 211588/196900; st 37; 5.1: cur:opening tgt:open cl:0 op:-1
8102324405 mg_rpc.c:314            Shelly.SetState via WS_in 192.168.1.86:57995
8102329912 shelly_output.cpp:64    Output 1: off -> on (GDO:close)
8102336507 shelly_hap_garage_d:309 GDO 1: Tgt State: open -> closed (0 -> 1) (RPC)
8102342956 shelly_hap_garage_d:224 GDO 1: Cur State: opening -> stopped (2 -> 4)
8102645113 shelly_output.cpp:64    Output 1: on -> off (pulse_off)
8107391845 shelly_main.cpp:449     Up 8107.38, HAP 0/1/16 ns 1, RAM: 211588/196900; st 37; 5.1: cur:stopped tgt:closed cl:0 op:-1
```

Log  shows Target State from open -> closed at 8102336507, but Current state is stopped at 8102342956

The Shelly input has a input reed/magentic sensor to actually set the state to closed. 
In my opinion, this should be the source of truth if the garage door is indeed closed and set the target/current states accordingly to closed. 

Furthermore, The movement timeout period should also be utilised for this single sensor scenario the same way it operates for "Opening".
ie, if status is "Closing", but exceeds movement timeout, and input does not indicate closed. Set status Homekit status as "Open".

timoschilling, 2023-11-09:
That with the different states in the Home App is a UI bug there. We are not responsible for the UI, we only send a target state and a current state (`cur:stopped tgt:closed`). That state matches with what you said. I test your case with the "HomeKit Accessory Simulator" App and it looks like the Home App can't deal with the stopped state.

What you said about the Movement Time setting is right, but if you stopped the open or close process the Movement Time setting no longer matters because the Shelly knows that it is not closed and not (fully) open.

DanielNagy, 2023-11-09:
Hi @timoschilling. Thanks for your reply. 
I Understand you are not responsible for the Apple UI.

To add more UI confusion, Carplay shows Stopped.
[image]

But my callout was more, if its stopped, its technically open. rather than closed in the target state? no?

timoschilling, 2023-11-10:
The target state shows the direction of the last movement and that was close. What the UI does it's not our thing! We will not provide wrong informations to HomeKit just to fix UI Bugs. We does what the HomeKit specification says. If you want to get that bug fixed file an issue at Apple.

DanielNagy, 2023-11-10:
The last action was open. Not close.

markirb, 2023-11-10:
I do think there really is a bug @timoschilling:
Looking at the code this really should read stopped... 

https://github.com/mongoose-os-apps/shelly-homekit/blob/ceb8cb37d98cc8649fae69b3bdc45eb2866dabff/src/shelly_hap_garage_door_opener.cpp#L270C13-L270C36

I think i noticed that also with my garage door....
but this depends really on the garage controller whats correct here right? Hitting toggle on Opening is stopping the action at least on my door...

DanielNagy, 2023-11-13:
Hi @timoschilling.

So i had a play with the codebase over the weekend locally with my door. 
What the codebase logic doesn't account for is in a single sensor configuration, if the motor was obstructed whilst closing, It will stops and reverses to fully open position. 
Based on this scenario, the "Stopped" state should not be used here. The Final state (post movement timeout) should either be open, or closed purely based on the "Close sensor's" state.

The only part that needs attention is the Obstruction logic is fired, and then cleared because it was set to an open state.  That fix could be done with an additional obstruction_closing_detected_ flag or something.

This is the code i've added and tested inside the KClosing state (inside void GarageDoorOpener::RunOnce()) in the event the movement time expires when closing.

[image]

This is how my remootio 3 behaves with its single sensor to report its status (irrelavent of HomeKit).

d4rkmen, 2023-12-11:
As far as can see, the problem here is logic. As far as we not consider any obstruction sensor, the stopped (current) state could be set only when switching directions. Like opening->close and closing->open. Just to notify controller the movenent was stopped. Then we can raise next state immediately to continue movement in opposit direction OR stay stopped for the case door controller actually supports stopping. This behaivour could be defined by configuration (as new feature)

markirb, 2023-12-30:
Yes, exactly that would be a good solution to have that configurable.

I have a Hörmann Liftronic that behaves as

open / stop / close

maybe we can gather info about different manufacturers?

closed 2024-01-29 by github-actions[bot]

tomh79, 2024-07-10:
> Shelly Plus 1, with single sensor for Closed status.
>
> To replicate: When Garage status is closed, press "Toggle" from the WebGUI to open door, then press "Toggle" again after a few seconds (but [...]

Hello Daniel, im currently using the Plus 1 (homekit flashed) to open/close my garage. Works perfectly via Homekit on iphone (momentary switch). I'm trying to build in a door status but cannot find any info which sensor to use and how to implement it in this setup? Could you explain how you set it up? Thanks!