mongoose-os-apps/shelly-homekit #1336: Homekit Switch Type

GaryGlitHub, 2023-11-11:
The Homekit Accessory Protocol offers a selection of input and output types.
The Mongoose firmware allows us to select one of the options. Would it be possible to expand the available options? The switch types especially... Homekit offers specific options only available to the type of switch.
The stateless switches lack the possibilities of automations whereas the normal toggle switches do offer these.

[image]

The workaround would be to create a Homekit Shortcut, but these operate really slow.
A switch with a Shortcut attached to it suffers from a 2 - 3 sec delay. This seems surmountable, but in practice people think a switch is dead if it does not respond in almost instantaneously and they start to operate the switches wildly.
Anyway, if it's possible to add the HAP switch type inside the mongoose firmware, it creates the opportunity for faster automations.

rionshin, 2023-11-11:
If you have 2-3 sec delay on Shortcuts  - there is something wrong with your Apple hub or network, my shortcut delay is around 150ms from the moment of the execution to the action, i can see it because all my shortcuts write in InfluxDB and i see all time stamps. Shelly I4 is really fast with shortcuts  - i have coded some shortcuts with more than 10 IF/Else inside checking various things before execute and never have more than half a second delay.  
Having Switch as HAP type mean something is directly commanded which is not the case for I4, but i can see the benefit to have Fake switch that triggers actions.

GaryGlitHub, 2023-11-11:
I really like the I4, as it is quite versatile. When it works properly in homekit it offers 4 programmable homekit actions (including shortcuts!) Plenty of reason to figure out if I can shorten delay. I am quite certain the I4 is not the issue here, but rather the Homekit implementation in conjunction with Shortcuts.
I also have a couple of Shelly I3's installed behind toggle switches. I changed those to 'Toggle, on = single, off = double'. The single and double press send there commands directly and the result is good. No serious delays...

rionshin, 2023-11-11:
Do you use buttons for the I4 -  because it offer 12 actions not 4... and i think that's your problem...   That's one of the I4  Inputs.  and shortcut behind long press
[image]
[image]

GaryGlitHub, 2023-11-12:
You are right: the i4 has 12 options: the 4 inputs have 3 options each. I tried two Shelly's for the same room: one is the Shelly i4, the other the 1PM. The 1PM has automations that create a toggle switch, the i4 needs a shortcut to create that.
The 1PM is instant, the i4 shortcuts are not.

timoschilling, 2023-11-13:
@GaryGlitHub I have the 2-3 sec delay for shortcut actions as well. So I know the problem but I can't see how a "fake stateful" switch will fix that. What are the automatisations you miss with the stateless switch?

@rionshin what kind is your connected HomeHub?

GaryGlitHub, 2023-11-15:
Ik solved it! 
Homebridge has helped me out: https://github.com/eduardoboucas/homebridge-toggle-switch

closed 2023-11-16 by timoschilling