mongoose-os-apps/shelly-homekit #1338: Cannot get wifi established after a couple of weeks...

tkt2rde, 2023-11-16:
I have two Shelly Plus 1's connected to ceiling fixtures.  I updated to latest firmware, then did the HomeKit flash and everything went perfect.  After working for few weeks, I started having issues with them showing not connected in Homekit and in the Shelly App.  I tried reserving IP's and rebooting router, but for whatever reason, they just stopped connecting to my wifi network.  I have held down button on unit to reset, tried the reset option in the GUI connected directly via AP, but I'm not having any luck.  While connected via Shelly AP, everything responds normally, but it just times out trying to connect every time I attempt on either WIFI 1, or WIFI 2, reverting back to AP after every failure.  I was hoping there was an update or something, but I cannot connect via wifi, so I cannot use the same process I did initially to try reinstalling.    I started the logger, it gives a reason 201, but I have no clue what that means.

`3050453468 shelly_debug.cpp:231    No log file, sending new entries
3056612249 shelly_main.cpp:449     Up 3056.60, HAP 0/0/16 ns 0, RAM: 211812/203988; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3058070681 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3059072689 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3060073723 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3061074406 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3062092917 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3064080502 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3064611410 shelly_main.cpp:449     Up 3064.60, HAP 0/0/16 ns 0, RAM: 212044/203988; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3065090752 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3066080180 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3067081005 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3068080538 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3069080994 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3070091092 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3071091138 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3072089976 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3072611255 shelly_main.cpp:449     Up 3072.60, HAP 0/0/16 ns 0, RAM: 212044/203988; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3073083742 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3074082015 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3075086421 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3076083190 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3077085710 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3078085986 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3079085488 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3080087393 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3080611290 shelly_main.cpp:449     Up 3080.60, HAP 0/0/16 ns 0, RAM: 212044/203988; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3081083504 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3082086214 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3083086665 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3084086214 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3085086019 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3086091595 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3087085621 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3088086110 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3088611225 shelly_main.cpp:449     Up 3088.60, HAP 0/0/16 ns 0, RAM: 212044/203988; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3089083227 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3090087886 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3090767483 mg_rpc.c:314            Shelly.SetWifiConfig via WS_in 192.168.33.2:60978
3090793521 shelly_wifi_config_:134 New config: {"ap": {"enable": false, "ssid": "ShellyPlus1-B8D61A8C55D4", "pass": ""}, "sta": {"enable": true, "ssid": "IMAGINE", "pass": "************", "ip": "", "netmask": "", "gw": ""}, "sta1": {"enable": false, "ssid": "", "pass": "", "ip": "", "netmask": "", "gw": ""}, "sta_ps_mode": 1} 1 0
3090803417 shelly_wifi_config_:140 Setting ps mode to 1
3090835025 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3091090469 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3092089841 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:60978
3094606799 shelly_wifi_config_:264 STA  config: 1 IMAGINE ************
3094610773 shelly_wifi_config_:275 STA1 config: 0  
3094620704 I (3095245) wifi:mode : sta (b8:d6:1a:8c:55:d4) + softAP (b8:d6:1a:8c:55:d5)
3094623648 I (3095255) wifi:enable tsf
3094636402 E (3095265) wifi:AP has neither DSSS parameter nor HT Information, drop it
3094650702 E (3095275) wifi:AP has neither DSSS parameter nor HT Information, drop it
3094718368 E (3095345) wifi:AP has neither DSSS parameter nor HT Information, drop it
3094746165 E (3095375) wifi:AP has neither DSSS parameter nor HT Information, drop it
3094755280 E (3095385) wifi:AP has neither DSSS parameter nor HT Information, drop it
3094762511 E (3095395) wifi:AP has neither DSSS parameter nor HT Information, drop it
3095245087 E (3095875) wifi:AP has neither DSSS parameter nor HT Information, drop it
3095417288 E (3096045) wifi:AP has neither DSSS parameter nor HT Information, drop it
3095728829 mgos_mongoose.c:66      New heap free LWM: 203616
3095882014 E (3096515) wifi:AP has neither DSSS parameter nor HT Information, drop it
3095888293 mgos_mongoose.c:66      New heap free LWM: 202924
3096611389 shelly_main.cpp:449     Up 3096.60, HAP 0/0/16 ns 0, RAM: 204796/202924; st 53; 0.1: st:1 in_st:0 inm:1 ininv:0
3096789309 mgos_mongoose.c:66      New heap free LWM: 202856
3097165989 mgos_wifi_sta.c:621     Trying IMAGINE cfg 0 AP (auto) att 1
3097171912 esp32xx_wifi.c:359      WiFi STA: protocol BGN (0x7)
3097175580 I (3097805) wifi:Set ps type: 1
3097175846 
3097182546 mgos_net.c:89           WiFi STA: connecting
3097896537 E (3098525) wifi:AP has neither DSSS parameter nor HT Information, drop it
3097981869 E (3098615) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... 77 lines cut ...]
3124162912 E (3124795) wifi:AP has neither DSSS parameter nor HT Information, drop it
3124553292 E (3125185) wifi:AP has neither DSSS parameter nor HT Information, drop it
3124581319 mgos_wifi_sta.c:621     Trying IMAGINE cfg 0 AP (auto) att 4
3124586915 esp32xx_wifi.c:359      WiFi STA: protocol BGN (0x7)
3124590407 I (3125225) wifi:Set ps type: 1
3124590651 
3124597978 mgos_net.c:89           WiFi STA: connecting
3125440148 E (3126065) wifi:AP has neither DSSS parameter nor HT Information, drop it
3125485716 E (3126115) wifi:AP has neither DSSS parameter nor HT Information, drop it
3125571644 E (3126205) wifi:AP has neither DSSS parameter nor HT Information, drop it
3125604876 shelly_wifi_config_:306 Connection failed
3125628283 shelly_wifi_config_:310 Reverting to previous config: {"ap": {"enable": false, "ssid": "ShellyPlus1-B8D61A8C55D4", "pass": ""}, "sta": {"enable": false, "ssid": "", "pass": "", "ip": "", "netmask": "", "gw": ""}, "sta1": {"enable": false, "ssid": "", "pass": "", "ip": "", "netmask": "", "gw": ""}, "sta_ps_mode": 0}
3125749709 E (3126375) wifi:AP has neither DSSS parameter nor HT Information, drop it
3126750109 E (3127375) wifi:AP has neither DSSS parameter nor HT Information, drop it
3126757151 E (3127385) wifi:AP has neither DSSS parameter nor HT Information, drop it
3127141461 E (3127775) wifi:AP has neither DSSS parameter nor HT Information, drop it
3127163717 E (3127795) wifi:AP has neither DSSS parameter nor HT Information, drop it
3127170664 mgos_wifi.c:71          WiFi STA: Disconnected, reason: 201
3127174393 mgos_wifi_sta.c:638     Connect failed
3127179896 mgos_net.c:84           WiFi STA: disconnected
3128611450 shelly_main.cpp:449     Up 3128.60, HAP 0/0/16 ns 0, RAM: 209684/198120; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0
3128616324 shelly_wifi_config_:264 STA  config: 0  
3128620114 shelly_wifi_config_:275 STA1 config: 0  
3128625558 I (3129255) wifi:mode : softAP (b8:d6:1a:8c:55:d5)
3136611255 shelly_main.cpp:449     Up 3136.60, HAP 0/0/16 ns 0, RAM: 211372/198120; st 52; 0.1: st:1 in_st:0 inm:1 ininv:0`

salemsayed, 2023-11-28:
Same problem here on 2 different Shelly1 devices, just happened out of nowhere, there was no change in any of the network settings

tkt2rde, 2023-11-29:
Hopefully someone knowledgeable can help.   Dead in the water for now.

haralddvorak, 2023-12-15:
Had the same problem. Solved it by hard resetting the shelly (bridging L1 and SW five times within 60 seconds after power connection).

closed 2024-01-13 by github-actions[bot]