mongoose-os-apps/shelly-homekit #1347: Shelly firmware v1.1.0 doesn't update

jnnsrctr, 2023-12-19:
I don't know if this is an issue with the Shelly original fw (I guess so), but I was not able to flash a Shelly Plus 1 that I updated to v1.1.0 before. Downgrading to v1.0.8 did the trick.

lucassite120, 2023-12-20:
Please, how I can downgrade my Shelly to v1.0.8?

mrfreon, 2023-12-21:
I have the same issue, I cannot flash the Shelley Plus 1 on V1.1.0. Can anyone help with the downgrade to v1.0.8 ?

Stoppersp, 2023-12-21:
@mrfreon From what I can find online, it appears that downgrading 2nd gen or plus devices isn't possible. I am in stuck in the same situation. Upgraded the firmware on my shelly plus 1 before trying to flash mongoose and it wont work with 1.1.0 and no way that I can find to downgrade the firmware.

jnnsrctr, 2023-12-21:
I got you...

v1.0.8 is attached
[Plus1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/13739814/Plus1.zip)

mrfreon, 2023-12-21:
@jnnsrctr - Thanks! I am a bit of a novice with this stuff, How do I downgrade the device with your file?

jnnsrctr, 2023-12-21:
@mrfreon In the Shelly GUI, go to Settings > Firmware. You can upload it there, worked for me. Maybe restart the Shelly before.

mrfreon, 2023-12-21:
@jnnsrctr So I am in settings > Firmware and im dragging your file into the 'drag and drop to upload' box and nothing happens even after a shelly restart.

jnnsrctr, 2023-12-21:
Just tried it again, worked like a charm.
But you clicked the button, I hope?
[image]

mrfreon, 2023-12-21:
Give the man a beer! Thank You. 
Yes I had clicked the button.. I was trying to upload an unzipped file which it didn't like.
Im on V1.0.8 now

Stoppersp, 2023-12-21:
@jnnsrctr thanks for the firmware! I could not find it anywhere. You are a lifesaver!

Acidfreeze, 2023-12-22:
I am in the same situation but with a Shelly Plus1PM is there also a firmware downgrade File to Version 1.0.8

Thank you in advance

Areae51, 2023-12-22:
@jnnsrctr Can you please help me and @Acidfreeze we booth have the same problem.

lalolerry, 2023-12-23:
@jnnsrctr Me too havew this issue, via a Shelly plus 1PM stuck on 1,1,o stock firmware - no way to update to mongoose!

Shadow00000z, 2023-12-25:
> @mrfreon Just tried it again, worked like a charm. But you clicked the button, I hope? <img alt="Screenshot 2023-12-21 at 11 58 41" width="1328" src="https://private-user-images.githubusercontent.com/99033602/292181975-73d16ce4-7fe0-4a00-a45c-975089cd7d77.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM1NDE1MTYsIm5iZiI6MTcwMzU0MTIxNiwicGF0aCI6Ii85OTAzMzYwMi8yOTIxODE5NzUtNzNkMTZjZTQtN2ZlMC00YTAwLWE0NWMtOTc1MDg5Y2Q3ZDc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI1VDIxNTMzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwNzNiMzIyYWM1OWRjZTQwOTY2ZjA3ZTEwZDhkYjIyNmE0ZGU5YTJjNWJmYTU2NjRhNjNjZGEyZmVhNjBjNGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Q2W1HEAAZoE8N9isAGsj5aKr4qxqPaeVtRFvppbNpls">

this isnt working for me. no action when i click the button of the zipped firmware. tried different browsers too. any ideas how i can get this firmware to roll back? thanks!

Shadow00000z, 2023-12-25:
> I got you...
> v1.0.8 is attached
> [Plus1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/13739814/Plus1.zip)

+1 @jnnsrctr im stuck in 1.1.0 like the others with a plus1PM cannot downgrade with method above. Any ideas?

mrfreon, 2023-12-26:
@Shadow00000z 
If you are using a Mac, try holding down option and then download Plus1.zip. This worked for me.

Shadow00000z, 2023-12-26:
> @Shadow00000z
> If you are using a Mac, try holding down option and then download Plus1.zip. This worked for me.

@mrfreon i can dl the zip, but when I load it into the firmware file selection, but nothing happens when I click update

lucassite120, 2023-12-26:
Since there isn't any real update of this firmware, I'm moving all my Shellys to HAA firmware. It's a great discover, with many customization options: https://github.com/RavenSystem/esp-homekit-devices

Followed these steps: https://www.youtube.com/watch?v=tXkaTqY86cc

jnnsrctr, 2023-12-27:
Unfortunately, I cannot supply firmware for other Shellys. I only run Plus1 at the moment.

lalolerry, 2024-01-12:
maybe this solution can help you. look at my post here:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1351

simonfield76, 2024-01-15:
> I got you... v1.0.8 is attached [Plus1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/13739814/Plus1.zip)

Thank you this finally worked for me….. 

Updated after a hard reset worked like a charm!!!!

eddiebullman, 2024-01-30:
have down graded a Plus1 to v1.0.8 and done a factory reset and still will not update firmware

eddiebullman, 2024-01-30:
have downgraded to v0.14.1 and still will not update

jnnsrctr, 2024-01-30:
@eddiebullman Most likely an unrelated issue on your side. Did you try different browsers? What's the problem exactly?

eddiebullman, 2024-01-30:
have tried updating from a Mac and a windows PC same issue just will not take new firmware

jnnsrctr, 2024-01-30:
@eddiebullman Thats not the most detailed error description. Did you download the zip and tried it that way, or only via OTA?

eddiebullman, 2024-01-30:
when it was on v1.0.8 tried OTA and tried uploading the file download from this link 
http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1.zip

Uploaded the zip file while on windows pc you get the progress bar and then the unit reboots but came back with stock firmware again

On v0.14.1 doesn't give the option of uploading zip file but you get a custom firmware section and I use the link above get progress bar unit reboots and the come back up with stock firmware

silverter, 2024-01-30:
> when it was on v1.0.8 tried OTA and tried uploading the file download from this link http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1.zip
>
> Uploaded the zip file while on windows pc you get [...]

My suggestion would be, as I posted before, to disconnect the device from power completely after the update. Download the respective zip firmware, upload it to the device and let the update process run. After the progress bar is complete and de device reboots, disconnect it from power.

eddiebullman, 2024-01-30:
> My suggestion would be, as I posted before, to disconnect the device from power completely after the update. Download the respective zip firmware, upload it to the device and let the update process [...]

will give it a try at some point but not tonight

closed 2024-02-28 by github-actions[bot]