mongoose-os-apps/shelly-homekit #1350: Shelly +1 with latest Firmware failes

gregorw, 2023-12-23:
The OTA flashing of a Shelly +1 with the latest firmware 20231219-133950/1.1.0-g34b5d4f fails with the following error:

``` 
1703368784.279 2 shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_progress", "msg":"in_progress", "progress_percent":68, "ts":1703368784.19}
1703368785.356 2 shelly_ota.cpp:403      Plus1-2.11.2/bootloader.bin: 23728 bytes (hdr 57 desc 0 crc fd046e7a)
1703368785.375 0 shelly_ota.cpp:311      Aborting update: backend 0x3ffe2ba4 aborts update at file Plus1-2.11.2/bootloader.bin: -4: app / fs already written, can't update boot
1703368785.385 2 shelly_ota.cpp:234      OTA status error, 69 %
1703368785.410 2 shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1703368785.39}
1703368785.425 2 shelly_http_client.:606 0x3ffe2360: Finished; bytes 1150560, code 200, redir 0/3, auth 0, status CANCELLED: Cancelled
```

silverter, 2023-12-26:
see my comment on  #1351  - You might give it a try and downgrade to 0.14.x.

gregorw, 2023-12-28:
> see my comment on #1351 - You might give it a try and downgrade to 0.14.x.

Thank you. The firmware 1.0.8 from https://github.com/mongoose-os-apps/shelly-homekit/issues/1347#issuecomment-1865954629 worked for me.

closed 2023-12-28 by gregorw