mongoose-os-apps/shelly-homekit #1351: shelly plus 1PM stuck in 1.1.0

Shadow00000z, 2023-12-25:
please help me downgrade. i am stuck and the buttons take no action when trying to downgrade firmware.

anyone have ideas?

silverter, 2023-12-26:
I have the same issue. I cannot downgrade nor flash any other firmware on the device. 
Any ideas ?

silverter, 2023-12-26:
My positive Update on this one: 
I was able, after many attempts to downgrade the 1.1.0 firmware to 0.14.x from the following Link: https://www.shelly-support.eu/forum/thread/21566-firmware-1-0-0-für-geräte-der-2-generation-wird-offiziell-bereitgestellt/?postID=227817#post227817
After that, I was abler to refresh the device with the Homekit firmware. 
Hope this helps.

Shadow00000z, 2023-12-26:
> My positive Update on this one:
> I was able, after many attempts to downgrade the 1.1.0 firmware to 0.14.x from the following Link: [...]

@silverter i may be going crazy but I don’t see a file to download or url to firmware via the link in your post.

silverter, 2023-12-26:
> @silverter i may be going crazy but I don’t see a file to download or url to firmware via the link in your post.

This is the direct link to the 0.14.x. Firmware for the Shelly Plus 1PM: https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1PM/fw-signed.zip

ThomasM60, 2023-12-31:
Unfortunately this downgrade link is not the solution (anymore?). I tried to downgrade  my Shelly 1 Plus.
After passing the update procedure and reconnecting it is still firmware v1.1.0. 
Either firmware v1.1.0. prevents any downgrade or Shelly provides now the newest firmware under the following link:

https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1/fw-signed.zip

silverter, 2024-01-04:
> Unfortunately this downgrade link is not the solution (anymore?). I tried to downgrade my Shelly 1 Plus. After passing the update procedure and reconnecting it is still firmware v1.1.0. It seems that Shelly provides now the newest firmware under the following link:
> 
> https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1/fw-signed.zip
> 
> But, thanks to #1347, I could downgrade my Shelly 1 Plus to firmware V1.0.8. This firmware still supports the update to the respective Apple Home firmware

I had that same issue, that why I wrote after many attempts. What I think help me what go through the downgrade process, then disconnect the device from power. After the reconnection to power, I had the 0.14.x version on the device. You might give it a try.. it worked for my 2 devices. 
Also, do not attempt to downgrade using the web-link. Rather download the firmware and load it onto the device. Good luck.

lalolerry, 2024-01-12:
@silverter YES! it worked! thanks a lot!

I made what you have suggested to downgreade a shelly Plus 1PM stuck in  firmware version 1.1.0 to 0.14.1 firmware version, and thank flashed via OTA to Shelly homeklit  2.12.2 firmware.
This is step by step what I made:

1. in firmware section of web UI I pasted the link to Plus 1PM 0.14.x firmware provided by markirb (thanks!) in an other post, form this one official repository:
https://www.shelly-support.eu/forum/thread/21939-downgrade-links-to-0-14-x/
(see screenshot below)
[image]

2. clicked on update form link button, and wait - Shelly device  led start flashing, indicating thst it was really dowloading a firmware on device (see screenshot below)
[image]

3 When the updating firmware process bar showed 95%, and right after the Shelly device  led stopped blinking,  I POWERED OFF THE DEVICE cutting the main power to Live and neutral entry. 
NOTE: the updating firmware process bar MUST NOT be at 100% level or the shelly device turns on with 1.1.0 firmware again. Anyway you can start the updating porcess again and try again.
(see screenshot below)
[image]

4. I gave power back to the shelly plus 1PM device and... it powered up in acess point mode, because it lost all settings... but RUNNING 0.14.1 FIRMWARE!

5. I inserted wifi netwirk name and password as usual and reconnet it to my wifi
(see screenshot below)
[image]

7. in firmware section pasted the link to shelly homekit firmware, this one:
http://shelly.rojer.cloud/update
and clicked on update form link button
(see screenshot below)
[image]

8. And finally I had the homekit firmware running on my Shelly Plus 1PM device!
(see screenshot below)
[image]

NOTE !1 TRY AT YOUR OWN RISK (AS I DID ;) )
It worked flowless on my 3 shelly plus devices stuck on 1.1.0 firmware - the hint is to power off the devices just at the end of firmware downloading and right before it restart with the already installed 1.1.0 firmware if you don't power it off. And onestrly i don't think you can brick the device if you power off in the worng moment, because of safety checks that shelly device makes before running a firmware (can't run a corrupted file).

NOTE 2: I THINK this trick should work with any shelly device stuck on 1.1.0 fw (just copy&past the right firmware link! form firmware section)... in the waiting of official updating fix fot this fiwmare that I think Allterco itself will/shell make in the near future

NOTE 3: I really love the shelly homekit firmware and I have about 50 shelly devices running it without any problem for years... how I wish it would still be developed and maybe ported to the new mini an plus devices! 

BR

Lalo

Jazz7584, 2024-01-20:
@lalolerry
Thx your solution worked for me.

martindavid-net, 2024-01-22:
does not work for me. I guess the "new" shelly that come with a strange built firmware dont work

Lubinko, 2024-01-25:
work for me .. thanks a lot!

Tyler1456, 2024-02-02:
Note, you DO NOT need to pull the power.  You can just quickly click the "Reboot" button when you're ~95% through the downgrade.  I'm glad this worked out for me, since my Shelly is already installed inside of a junction box. And...I didn't feel like flipping breakers.

z0lle, 2024-02-16:
@lalolerry Ive tried this methood 20 times now trying to time it.
Tried with rebooting, or cutting mains power compleetly, 
Hitting it exactly when 95% comes up, hitting it as soon as it stops blinking.
Hitting it at 90%.
i just cant get it to work.
Im on 1.2.0
When i got my device, i updated it like one of the first things i did.
I should have checked about this before hand.... So annoying!
Not sure what version i had, but i just cant get it to downgrade what so ever!

lalolerry, 2024-02-16:
> @lalolerry Ive tried this methood 20 times now trying to time it. Tried with rebooting, or cutting mains power compleetly, Hitting it exactly when 95% comes up, hitting it as soon as it stops [...]

Hi z0lle,
maybe the problem is original fw 1.2.0 (I had 1.1.0)?
Anyway, the correct timing is the clue for "my" method: just after the device led stop blinking (meaning it has finished downloading the old firmware on the device)... this is the right time to power it off (or click reboot as Tyler1456 said).
Try it!

LAst solution, flash it with old firmare non via OTA but via UART and PINOUT. Not so easy I know.

badewanne1234, 2024-02-17:
The 1.2.0 firmware is the absolute worst that have seen on a Shelly Plus 1PM! Loosing network connection after a couple of hours all the time, no problem until the 1.1.0 FW. 
Will try to downgrade today, thanks for the link with the older FW :)

Theo16, 2024-02-18:
> The 1.2.0 firmware is the absolute worst that I have seen on a Shelly 1PM! Loosing network connection after a couple of hours all the time, no problem until the 1.1.0 FW. Will try to downgrade today, thanks for the link with the older FW :)

I have the same problem with my 1PM and 2PM. Do You have any instruction how to go bach to 1.1.0?

badewanne1234, 2024-02-18:
> I have the same problem with my 1PM and 2PM. Do You have any instruction how to go back to 1.1.0?

There is no list of available FWs for Gen2 devices, 1.10 is not available, but for 1PM you can find the 1.08 FW here: https://github.com/mongoose-os-apps/shelly-homekit/issues/1347#issuecomment-1865954629

For 2PM, you can find 0.14 FW (its from 2023, dont get fooled about the version naming) here: https://www.shelly-support.eu/forum/thread/21939-downgrade-links-to-0-14-x/

To downgrade, login to the shelly, go to settings, firmware and upload the zip. 
In my case 1PM, there was no progress bar, it looked like nothing was happening when I pressed the upload button, but it was downgrading in the background without any information...

Good luck

TheGHubBoy, 2024-03-01:
I also had this issue, but my Shelly +1PM was on firmware version _1.2.2_ and couldn't downgrade to 0.14.1.
Following error was in the logs: `aborts update at file bootloader.bin: -4: app / fs already written can't update boot`
The explanation behind this error is, the _app / fs_ gets written before the bootloader due to different order in the zip file.

-> So I edited the zip file and tried to update / downgrade the Shelly and surprisingly it worked.

Don't blame me if your Shelly is bricked afterwards.

I apologize for my broken or not perfect english, since english is not my first language!

Down below I have attached the update file I used to update my Shelly.
**Again: Use this update file at your own risk!**

[fw-edited.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/14466540/fw-edited.zip)

z0lle, 2024-03-03:
> I also had this issue, but my Shelly +1PM was on firmware version _1.2.2_ and couldn't downgrade to 0.14.1. Following error was in the logs: `aborts update at file bootloader.bin: -4: app / fs already written can't update boot` The explanation behind this error is, the _app / fs_ gets written before the bootloader due to different order in the zip file.
> 
> -> So I edited the zip file and tried to update / downgrade the Shelly and surprisingly it worked.
> 
> Don't blame me if your Shelly is bricked afterwards.
> 
> I apologize for my broken or not perfect english, since english is not my first language!
> 
> If you encounter any problems, maybe try sharing your log files here? Then maybe somebody can help?
> 
> Down below I have attached the update file I used to update my Shelly +1PM. **Again: Use this update file at your own risk!**
> 
> [fw-edited-plus1PM.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/14466540/fw-edited.zip)

Can you explain more detailed what is changed in this file?

TheGHubBoy, 2024-03-03:
> Can you explain more detailed what is changed in this file?

Sure I can try.

I was trying to flash the old firmware (0.14.1) and got this error in the logs: 
`aborts update at file bootloader.bin: -4: app / fs already written can't update boot`

So I did a quick google search and found this issue on the Tasmota GitHub: 
[tasmota/mgos32-to-tasmota32#31 (comment)](https://github.com/tasmota/mgos32-to-tasmota32/issues/31#issuecomment-1866001196)
(User described the same problem)

This User also mentioned how to fix this error:
> Pack files into ZIP in this order and all will work fine again:
 
[tasmota/mgos32-to-tasmota32#31 (comment)](https://github.com/tasmota/mgos32-to-tasmota32/issues/31#issuecomment-1866559748)

So I opened WinRAR and drag and dropped the files one by one just like they were specified in the manifest.json and uploaded the file into the Shelly Web Interface. Unfortunately that didn't work, I got following error 
`zip compression is not supported`

I scrolled down a bit and found this 
> As I understand it, the compression method must be "store". On most zip-applications the default is "deflate".

[tasmota/mgos32-to-tasmota32#31 (comment)](https://github.com/tasmota/mgos32-to-tasmota32/issues/31#issuecomment-1866420285)

So I drag and dropped the files one by one into WinRAR again, but this time specified that the files compression method should be "store".
Uploaded the file into the Shelly Web UI and the update / downgrade succeeded.

I hope that helps.

Sorry if this is hard to read/understand since English is not my first language!

iStefo, 2024-03-03:
@TheGHubBoy thanks for your explanation (which is understandable perfectly)  and the bravery to try it out :)

It can confirm that downgrading to 0.14 using your fixed FW file worked for me! Afterwards, I was able to upgrade to the HomeKit firmware using the regular OTA process.

copperfog, 2024-03-05:
@TheGHubBoy Thank you! This has been the only solution to work for me.

lucassite120, 2024-03-07:
Because this firmware will not be updated anymore, I'm using HAA firmware that works with Plus 1PM. I followed these steps to flash Shelly for Homekit: https://www.youtube.com/watch?v=tXkaTqY86cc

z0lle, 2024-03-08:
@TheGHubBoy 

> I also had this issue, but my Shelly +1PM was on firmware version _1.2.2_ and couldn't downgrade to 0.14.1. Following error was in the logs: `aborts update at file bootloader.bin: -4: app / fs [...]

That worked beuatifully for me :)
Thanks a lot for the help!

Now the next issue is, where do i find the powerconsumption in the homekit app? ^^

Fialky92, 2024-03-24:
Edited fw worked for me too. Thank you 🫶🏻

lucassite120, 2024-04-08:
If you want, you can try HAA firm. I flashed my Shelly Plus 1PM and it works like a charm. There is a new video with subs explaining all:
https://www.youtube.com/watch?v=06YHkRkwJE4

markirb, 2024-05-14:
I also tried to fix this with the posted findings for this fw:

https://github.com/markirb/shelly-homekit/tree/gh-ci

@timoschilling can you please verify?

markirb, 2024-05-18:
For all of you having this problem: Please try recent beta1 in "Releases Section"

markirb, 2024-06-04:
I am closing this issue, please use https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta2 until next official release

closed 2024-06-04 by markirb