mongoose-os-apps/shelly-homekit #1352: Shelly Plus i4 - not updating to HomeKit FW

kwn83, 2023-12-26:
Hi
Following instructions here:
https://github.com/mongoose-os-apps/shelly-homekit

Specifically run the following command:
http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlusI4.zip 

Command seems to just update to 1.10 (Shelly native) and thus does not update to be HomeKit enabled. 

What should I do/try?

Cheers

markirb, 2023-12-28:
See similar issues, e.g. here:
https://github.com/mongoose-os-apps/shelly-homekit/issues/1350

Problems seems that the newer version of shelly do not allow to update the bootpartition.
This seems to be the problem for all 2nd Gen/Plus devices.
According to the issues, a downgrade of the stock fw is a workaround.

It is not clear if another workaround is currently possible for this fw...

raphafein, 2024-01-07:
Hi,
same Problem here :/
Is there a firmware-download for the shelly plus i4, that can be used to downgrade?
I searched the web but without luck so far :(
Thanks in advance for any help!

markirb, 2024-01-09:
https://www.shelly-support.eu/forum/thread/21939-downgrade-links-to-0-14-x/

?

raphafein, 2024-01-10:
> https://www.shelly-support.eu/forum/thread/21939-downgrade-links-to-0-14-x/
> 
> ?

Hi @markirb and thanks for the hint. I wasn’t specific enough - sorry. I already tried this version but failed to downgrade the device.
The update process starts but the firmware version of my i4 stays 1.1.0 (20231219-133933/1.1.0-g34b5d4f). There was no success or error message. I thought it might be a problem with the downgrade link but maybe its a bug with the 1.1.0 firmware :/

markirb, 2024-01-12:
No I suspect the downgrade has the same issue as flashing of this Firmware. Do you have any logs?

lalolerry, 2024-01-12:
> Hi @markirb and thanks for the hint. I wasn’t specific enough - sorry. I already tried this version but failed to downgrade the device. The update process starts but the firmware version of my i4 [...]

maybe this solution can help you. look at my post here:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1351

martindavid-net, 2024-01-17:
have the same problem. also the download-link for the 0.14 stable ... does not work. it simply does not downgrade, no errors and restart in 1.1.0

martindavid-net, 2024-01-22:
it seems that those shellys who come with the following firmware dont comply with the 0.14 downgrade. sucks.

Firmware version: i4prod0

Firmware build ID: 20211101-134319/i4prod0-g2345cff-279-i4-production-bundles-2143-dirty

Web build ID: 1.5.5-8cbd45f

closed 2024-02-21 by github-actions[bot]