mongoose-os-apps/shelly-homekit #1362: Unable to add button actions to a Stateless Switch

brsnik, 2024-01-26:
When I press the switch I'd like to be able to turn on a light connected via an Ikea Hub.

[image]

[image]

However I am not able to save Actions in Apple Home. Get the following errors:

[image]
[image]

timoschilling, 2024-01-26:
I expect that's a problem with your Home setup or with the Home app and not with the accessory.

What happen when you create other automatisations?

brsnik, 2024-01-26:
When I set the type to Contact Sensor, I was able to make an automation. But not with Stateless Switch. Unfortunately Contact Sensor does't solve my problem, and I really need Stateless Switch.

Might be a Home app issue/bug. IDK.
When you say problem with my home setup what do you mean?

brsnik, 2024-02-17:
Still having this issue...

closed 2024-03-18 by github-actions[bot]