mongoose-os-apps/shelly-homekit #1363: after shelly firmware update not possible to upadate to homekit with mongoose methode

polleke251, 2024-01-26:
after shelly firmware update not possible to upadate to homekit with mongoose methode

dd150999, 2024-01-28:
same here
getting this error:
shelly_ota.cpp:311 Aborting update: backend error while processing data: -1 (-10: Cannot relocate shelly)

g3ntleman, 2024-02-18:
I tried to install a Shelly plus1, which came with some 0.x.beta firmware and failed to update using the methods on the shelly-homekit homepage. I then did the update to the current stock 1.2.0 as offered by the Shelly Web-UI.
This also failed silently (upload seemed to go smoothly, but it came back with the 1.2.0 firmware again.

After consulting the issues here, I finally succeeded by manually finding and downloading the **stock firmware 0.14**. 
I was then able to successfully install the curreny shelly-homekit FW.

I never saw any error messages during the update-process (in Firefox).

shaynemorgan, 2024-02-23:
Hi, can you share where you found 0.14 please? I hate the same issue.

woerchri, 2024-03-01:
http://192.168.0.159/ota?url=https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1/fw-signed.zip

worked for me. this way you are downgrading to 0.14 firmware which in turn let's you flash it for the shelly-homekit firmware.

g3ntleman, 2024-03-23:
I now had the same issue again, but with a the 1PM model. I updated to official FW 1.2.2 as recommended. Now it fails to downgrade to 0.14 as before. Help?

closed 2024-04-22 by github-actions[bot]