mongoose-os-apps/shelly-homekit #1366: Unable to update firmware for Shelly +1PM

jceab, 2024-02-13:
Hi all,

I just got a Shelly +1PM to integrate it into Apple HomeKit. I am miserably failing with updating the firmware to the HomeKit compatible version. It is currently using the stock firmware 20240213-140531/1.2.0-gb1b9aa8. Any attempt to update it via http://a.b.c.d/ota?url=http://shelly.rojer.cloud/update or http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1PM.zip (of course A.B.C.D replaced by the device's local IP) just ends in "Trying to update from http://shelly.rojer.cloud/update" but nothing is happening at all.

I've tried this with different browsers: safari, chrome, edge, brave... yet same result

May someone tell me what am I possibly doing wrong?

z0lle, 2024-02-16:
I have the exact same issue! It just gets stuck on trying to update.
Ive even tried downloading the file, and uploading it manually to the device, but it does not seem to be updating at all.

z0lle, 2024-02-16:
@pmkane Any chance you can take a look at this? Im not sure if its just for people running a Shelly + 1PM or more.
But a quick google seem to suggest at least that people running Shelly 1PM also get this issue.
Maybe its for all devices, or just with the 1PM family.
In any case, i and many others would greatly appreciate if you could take a look at this!

nicppereira, 2024-02-16:
Hey!

I was facing this exact issue on a +1, and I found this suggestion, which worked perfectly: 

https://www.reddit.com/r/shellycloud/comments/190a093/cant_flash_1_plus_with_homekit_firmware/

Basically revert to older stock firmware and only after try flashing HomeKit firmware. I suspect there may be some compatibility issues with the newer stock firmware versions, however I am not a contributor to the project, so take this with a grain of salt. Hopefully someone with more knowledge could shine a light onto this issue.

z0lle, 2024-02-17:
> Hey!
>
> I was facing this exact issue on a +1, and I found this suggestion, which worked perfectly:
>
> https://www.reddit.com/r/shellycloud/comments/190a093/cant_flash_1_plus_with_homekit_firmware/ [...]

Yeah ive tried something similar on my +1PM, but it just wont downgrade.
It goes through the whole process of uploading the firmware, but it is just the same firmware installed when its done.
Someone was also sugesting to abort at 95% and ive tried like 20 times, but just cant get it to stick.

If you have the link you showed, but for +1pm as a OTA link, i would gladly try that also, but the one in the thread is just for +1.

oli-d, 2024-02-20:
Hi! I have a Plus1 and am facing exactly the same issue. 

I am not even able to downgrade to the older official firmware as suggested in the reddit link. I see the same "Trying to update from..." message as for the Homekit firmware.

Also tried to upgrade from a ZIP file with no success. Just see an error in the browser console:

[image]

The only thing that works for me is to update the firmware to the latest stable or latest beta versions.

My suspicion is that they changed the way OTA updates work...

oli-d, 2024-02-20:
Update from my side. I finally managed to install the Homekit firmware 💪🏻🎉

The solution was described here: https://github.com/mongoose-os-apps/shelly-homekit/issues/1351#issuecomment-1872945621

What I did was:
* rebooted the device (obviously not sure whether this is required)
* Downloaded the Firmware 1.0.8. Got this from this post: https://github.com/mongoose-os-apps/shelly-homekit/issues/1347#issuecomment-1865954629
* Drag&Drop the ZIP file in the firmware screen and clicked "Update from file"

The system updated, restarted and was downgraded to 1.0.8. Running that firmware I was able to use the "classic" URL ```http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update``` to install the Homekit FW.

Hope that helps!

*Disclaimer*: this was for a Plus1, not Plus1PM!

copperfog, 2024-02-24:
Same here on Plus 1PM.

woerchri, 2024-03-01:
what helped me with my Shelly Plus 1 (mind the link is for the P1 but you can also find the 1PM firmware on the same server) was to downgrade to firmware 0.14: http://a.b.c.d/ota?url=https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1/fw-signed.zip

then I could update the shelly-homekit firmware: http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1.zip

z0lle, 2024-03-02:
> what helped me with my Shelly Plus 1 (mind the link is for the P1 but you can also find the 1PM firmware on the same server) was to downgrade to firmware 0.14: http://a.b.c.d/ota?url=https://shelly-infra1-eu.shelly.cloud/stable-firmware/Plus1/fw-signed.zip
> 
> then I could update the shelly-homekit firmware: http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1.zip
> 
> here is the link to the 0.14 firmwares: https://www.shelly-support.eu/forum/thread/21939-downgrade-links-to-0-14-x/

This is the files ive been trying with all along!

copperfog, 2024-03-05:
The 0.14 firmware downgrade isn't working with the +1PM either. People keep posting solutions then qualifying that it worked for their +1.

copperfog, 2024-03-05:
This solution finally worked for me: https://github.com/mongoose-os-apps/shelly-homekit/issues/1351#issuecomment-1974011119

ladolasso, 2024-03-29:
> Update from my side. I finally managed to install the Homekit firmware 💪🏻🎉
> 
> The solution was described here: [#1351 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/issues/1351#issuecomment-1872945621)
> 
> What I did was:
> 
> * rebooted the device (obviously not sure whether this is required)
> * Downloaded the Firmware 1.0.8. Got this from this post: [Shelly firmware v1.1.0 doesn't update #1347 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/issues/1347#issuecomment-1865954629)
> * Drag&Drop the ZIP file in the firmware screen and clicked "Update from file"
> 
> The system updated, restarted and was downgraded to 1.0.8. Running that firmware I was able to use the "classic" URL `http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update` to install the Homekit FW.
> 
> Hope that helps!
> 
> _Disclaimer_: this was for a Plus1, not Plus1PM!

this worked for me, thanks! homekit fw finally running

closed 2024-04-27 by github-actions[bot]