mongoose-os-apps/shelly-homekit #1370: Feature request/question: Support for Shelly Plus Uni

zdychacek, 2024-02-29:
Hi,

thank you for your work on this project. It's awesome, and I use your firmware daily in my light switches.

Now I am considering buying Shelly Plus Uni for controlling my entrance gate.

But I do not know if there is a support for Shelly Plus Uni (https://www.shelly.com/en-us/products/shop/shelly-plus-uni)? If not is it planned for near future?

Thank you for your response.

O.

nickmoone, 2024-05-16:
Any update on when support for the Shelly Plus Uni will be added? Isn't it just the same as the Shelly Uni with a different device ID?

markirb, 2024-05-21:
No it is not. It is an ESP32 device instead of ESP8266. 

But still it should not be too hard.

Just needs time and hardware...

uplyft, 2024-06-05:
I've got the hardware and I'm happy to help build this, if it wouldn't be too annoying for someone on the team to offer a bit of direction.

markirb, 2024-06-18:
Just compare pull requests that add a new model (plus plug s for example). Then add pins you find on other projects like tasmota. If you have questions just feel free to ask