mongoose-os-apps/shelly-homekit #1375: Shelly plus 2pm

xkyle22x, 2024-03-07:
Hi, is there any chance of Shelly plus 2pm being added to compatible devices now that the 2.5 is end of life or is it not possible?

lucassite120, 2024-03-07:
You can use HAA firmware. It works like a charm. I followed these steps to flash Shelly for Homekit and it worked from https://www.youtube.com/watch?v=tXkaTqY86cc

Alcor72, 2024-03-13:
> You can use HAA firmware. It works like a charm. I followed these steps to flash Shelly for Homekit and it worked from https://www.youtube.com/watch?v=tXkaTqY86cc

On the way to test (again 2 Shelly 2.5 dead…) this solution but there is no comparison in terms of simplicity….

Come on boys and girls working on this… we need you😜

Alcor72, 2024-03-16:
> On the way to test (again 2 Shelly 2.5 dead…) this solution but there is no comparison in terms of simplicity….
>
> Come on boys and girls working on this… we need you😜

First I tried the homebridge solution, but honestly there again no comparison to the simplicity of the native homekit firmware...
I am still struggling to have roller shutter as... roller shutter and not windows in home app...

j1gg3, 2024-03-18:
Highly appreciated as I need to exchange my 2.5 shellys with 2PM ones. 

The native Homekit support was so nice and worked for almost 2 years like a charm. I would donate to get this working :)

DB2FM, 2024-03-28:
I would donate, too. 
himeKit for 2PM really appreciated…

stefs72, 2024-04-18:
Please ad Shelly Plus 2PM to the compatible devices

lucassite120, 2024-04-18:
Now there is an easy way to install HAA Homekit firmware on Plus 2PM with mgos32toHAA.
See the new video: https://www.youtube.com/watch?v=06YHkRkwJE4

Alcor72, 2024-05-13:
Still no news on this?
I guess if the problem comes from the power metering part… we could be satisfied of the firmware with only up and down time manually set. 
of course the calibration on the 2.5 was really nice but I guess the same interface with only the up and down time to set would be welcomed as well on the Plus2PM…😜

rjealexander, 2024-05-19:
here too a kind request: having to replace all my 2.5's with 2PM's and I am a big fan of this native firmware. Simplicity and elegancy. It's worth a donation too.

markirb, 2024-05-20:
let's continue this here, I think we can get this running sometime soon
https://github.com/mongoose-os-apps/shelly-homekit/issues/1013

closed 2024-05-20 by markirb

Gaud1, 2024-08-18:
Please please maybe i could make a donation for it i have 8 shelly 2,5 one shelly 2 pm and maybe 2,5 was good with HomeKit like roller shutter and not Security like 2 pm ! It could be so cool to be able to use 2 pm like 2,5.