mongoose-os-apps/shelly-homekit #1379: Update Shelly 1 Mimi

mwkin, 2024-03-21:
Trying to flash a Shelly 1 Mini, fresh out of its box, to HomeKit. 

I can find the device from laptop with IP address, and log in. Attempted to update by accessing the URL from here. Response from device is "Trying to update from http://shelly.rojer.cloud/update" but nothing happens after that, at least not within 20 minutes. 

1. am I doing something wrong?
2. is the update not compatible with the Mini, which doesn't seem to be listed in the table of applicable devices? Or are there future plans?

thanks!

BlancoDeMallloca, 2024-03-25:
Shelly 1 Mini is not support (yet)

follow issue [1313](https://github.com/mongoose-os-apps/shelly-homekit/issues/1313) for updates

BlancoDeMallloca, 2024-03-25:
@timoschilling issue can be closed

mwkin, 2024-03-25:
Ok, thanks. Don’t spend any time on the Mini for me. 

mk 

Inside every old person is a young person wondering what happened.

lucassite120, 2024-04-08:
Try it with HAA. I flashed 2 Shelly Mini 1 gen3 and it works like a charm. You have a new video with subs explaining all process.
https://www.youtube.com/watch?v=06YHkRkwJE4

mwkin, 2024-04-08:
Sorry but what’s HAA?

mk 

Inside every old person is a young person wondering what happened.

lucassite120, 2024-04-11:
HAA is a firmware for ESP chips to convert devices to Homekit, with a lot of options and configurations.
https://github.com/RavenSystem/esp-homekit-devices/wiki

mwkin, 2024-04-11:
I will look into that, thanks!mkSent from Michael's iPhone

https://github.com/RavenSystem/esp-homekit-devices/wiki

closed 2024-05-10 by github-actions[bot]