mongoose-os-apps/shelly-homekit #1382: Unable to flash Shelly Plus i4

jeanlucverlhac, 2024-04-05:
Hello,

I'm unable to flash FW within Shelly Plus i4 : It seem to not be able to flash partition table (following case below) or boot loader. Any idea to help ?

Thanks
Jean-Luc

shelly_ota.cpp:245 Aborting update: backend 0x3ffe5aac aborts update at file partition-table.bin: -4: app / fs already written, can't update pt
17:45:28
shelly_ota.cpp:172 OTA status error, 99 %
17:45:28
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1712331928.25}
17:45:28
shelly_http_client.:606 0x3ffe3ea0: Finished; bytes 1972224, code 200, redir 0/3, auth 0, status CANCELLED: Cancelled

Sislik42, 2024-04-05:
I just got exactly the same error on Shelly Plus 1 few minutes ago.

Firmware ID: 20240223-141953/1.2.2-g7c39781
Web build ID: 240213111718-ce341dc8-HEAD

shelly_ota_source_ht:32 Update URL: http://rojer.me/files/shelly/shelly-homekit-ShellyPlus1.zip (0x3ffdff84)
...
shelly_ota.cpp:172      OTA status in_progress, 69 %
shelly_ota.cpp:355      Plus1-2.11.2/bootloader.bin: 23728 bytes (hdr 57 desc 0 CRC fd046e7a)
shelly_ota.cpp:245      Aborting update: backend 0x3ffdf868 aborts update at file bootloader.bin: -4: app / fs already written, can't update boot

Full log is attached
[diagnostics-shelly-plus1-debug-log-2.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/14887596/diagnostics-shelly-plus1-debug-log-2.txt)

jeanlucverlhac, 2024-04-06:
Hello 
As I powered it @5V, I guess it could be not enough to let the internal flash process to correctly erase the partition, thus I've just try with power @15V and I got the same issue :-(   I go on searching .... and let know if I find a solution.

lucassite120, 2024-04-06:
OTA flashing HAA Homekit firmware now is very easy:
https://github.com/RavenSystem/mgos32toHAA?tab=readme-ov-file#shelly-convert-to-haa

lucassite120, 2024-04-14:
You have a video with subs explaining all process of OTA flashing Shelly Plus with HAA Homekit firmware:
https://www.youtube.com/watch?v=06YHkRkwJE4

closed 2024-05-13 by github-actions[bot]

markirb, 2024-05-20:
Please use latest beta it works