mongoose-os-apps/shelly-homekit #1383: Connecting to a new wifi entirely from homekit

jsalexjs, 2024-04-23:
Hi, I'd like to sell a few of my 3d printed lights, and was thinking about using this and a shelly rgbw2. However, it doesn't seem to be possible to connect to this without first pre configuring the wifi on the shelly or using the ip interface. Would it be possible to add functionality where the shelly receives the users wifi intimation entirely through homekit? Maybe I'm missing something... thanks!

jsalexjs, 2024-04-23:
- I know there are prebuilt circuits that can do this, but I cant find one that just does white light. I don't want the device to identify as RGB when it an only be dimmed on and off.
-

closed 2024-05-23 by github-actions[bot]

timoschilling, 2024-05-23:
> Would it be possible to add functionality where the shelly receives the users wifi intimation entirely through homekit?
That only works for certified devices, not for open source firmware.