mongoose-os-apps/shelly-homekit #1386: HomeKit - shelly 1+ - impossible

ledoudou68, 2024-05-05:
bonjour,
j'ai voullu flashé mes deux shelly 1 + . Le premier a bien marché et j'ai eu le QR code a la fin. Pour le deuxieme je n'y arrive pas. je n'arriv epas a obtenir le qr code comme avant. 
[image]

yzalvov, 2024-05-09:
Same problem on Plus1PM for me

lucassite120, 2024-05-09:
You can install HAA Homekit firmware on Shelly Plus devices with mgos32toHAA.
See the video: https://www.youtube.com/watch?v=06YHkRkwJE4

markirb, 2024-05-14:
I tried to fix the problem
https://github.com/markirb/shelly-homekit/tree/gh-ci

If you feel adventurous you can try:
[shelly-homekit-ShellyPlus1PM.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/15312699/shelly-homekit-ShellyPlus1PM.zip)

 (this worked on my)

[shelly-homekit-ShellyPlus1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/15312701/shelly-homekit-ShellyPlus1.zip)

(not tested)

br

InventoryAndi, 2024-05-16:
Same for me. Seems that they did an update and now flashing does not work anymore. For the + series it is mot possible to downgrade so would be awesome if this could be fixed 🙏🏻

markirb, 2024-05-16:
@InventoryAndi did you try the zip file from above?

markirb, 2024-05-16:
I published a beta release under "Releases": 2.12.0-beta1 that should fix this. Feel free to try out

jazevedodesigner, 2024-05-16:
Same problem whit me, there is a solution for that?

markirb, 2024-05-16:
Did you try with 2.12.0-beta?

If so how do you do it?

Can you enable debug logging on Shelly fw and watch and post the log messages during upgrade please?

jazevedodesigner, 2024-05-17:
yes I try it.

I upload the new 2.12.0-beta, but when do start update, nothing happens.

Or do I have to update it first? Because there is a newer version on shelly?

Regards

markirb, 2024-05-18:
What is the filename? beta1 unfortunately contains zip within zip. Filename you upload has to be:

shelly-homekit-ShellyPlus1.zip

jazevedodesigner, 2024-05-18:
I will try, and let you now.
Thank you

closed 2024-05-20 by markirb

InventoryAndi, 2024-05-21:
Hi. I am sorry, I was busy the last days so I was not able to test it. But seems that it works for jazevedodesigner. How can I upload the downloaded file to my shelly device? I only know the OTA method.

markirb, 2024-05-21:
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta2 there is a drag and drop web if in 1.3 firmware section of shelly...

InventoryAndi, 2024-05-21:
Ok thanks. For me nothing happens. I renamed it as you wrote "shelly-homekit-ShellyPlus1.zip" and when I press Update from File, nothing happens...
[image]

sebw, 2024-05-26:
Told my cousin to get two Shelly Plus 1 (model SNSW-001X16EU) to control his garage doors. He updated from stock firmware version 1.0.7 to 1.3.2, released 3 days ago. From there, he couldn't upgrade to Mongoose. 

He was able to flash Mongoose on the second from stock firmware 1.0.7.

markirb, 2024-05-26:
> Ok thanks. For me nothing happens. I renamed it as you wrote "shelly-homekit-ShellyPlus1.zip" and when I press Update from File, nothing happens...
> 
> [image]
> 
> 

I did not say rename the file. It is called like that. If you do not have a file called like that then  there is soemthing wrong.

InventoryAndi, 2024-06-03:
Right, sorry. I tried to upload the unzipped file. The zip file works great. Thanks for the fantastic job! Will donate

markirb, 2024-06-04:
Thanks. If you want to donate to me personally, you can use my PayPal https://www.paypal.com/donate/?hosted_button_id=RVFA9G5VMXRX8 ; the official link currently goes to rojer, the original maintainer, who currently cannot work on the project anymore due to conflicts of interest

InventoryAndi, 2024-06-04:
Ah sorry, I have already done it via the official link. Next time I will do it directly to you