mongoose-os-apps/shelly-homekit #1389: Python HomeKit update script cannot find Shelly device on network...

thynkgit, 2024-05-19:
Hi,

I have been trying to upgrade my Shelly firmware to add HomeKit visibility. When I run the script Python cannot see my device, but it can be seen when I look at my router. I have been trying to give myself a crash course in Python to determine what the problem could be with the script.

Could someone please assist with finding the cause of the script not detecting my device?

Thanks!

markirb, 2024-05-20:
The thing you leave out: What is the device you are trying to discover/flash?

thynkgit, 2024-05-20:
My device is the Shelly Plus 1.

markirb, 2024-05-20:
Might be network problems. Why not flash it via Web?

thynkgit, 2024-05-21:
I tried that method way back when I first started the process. It didn't work for me, so I moved on to Python and dug in with that method. I understand more now, so I may have better results using the URL.

markirb, 2024-05-21:
Which firmware are you on? newer stock fws can only be upgraded from latest beta see release section

thynkgit, 2024-05-21:
I'm on version 1 of the stock firmware. I'm trying to update to the HomeKit firmware version 2.11.2.

thynkgit, 2024-05-21:
I first tried updating it from the device's web UI, but it didn't work.

markirb, 2024-05-21:
As said, only latest beta works: https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta2

thynkgit, 2024-05-21:
When I use the web (URL) method, what should the format of the URL be besides the device IP address and the file's directy location?

thynkgit, 2024-05-21:
Sorry... that was 'file's DIRECTORY location'.

thynkgit, 2024-05-21:
Okay. When you mentioned stock firmwares and beta, I thought you were referring to Shelly's firmware. I'll use the link that you sent.

mringwal, 2024-05-24:
Hi there. Just found this project and it looks like I'm in a similar situation. I have a Plus i4 and used the original web interface to update to the latest firmware (20240522-112836/1.3.2-g34c651b). 

I've tried:
- the update url (http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update) in the browser
- the file URL (http://shelly.rojer.cloud/update) in the web interface,
- the fs.img from 2.12.0-beta2 release in the file based up.
but nothing seems to happen.  

Back to Python. The script finds the i4 in the device scan, but then does nothing. Using its IP address, I get: Devices found: 0 Upgradeable: 0 Flashed: 0

markirb, 2024-05-24:
> Hi there. Just found this project and it looks like I'm in a similar situation. I have a Plus i4 and used the original web interface to update to the latest firmware [...]

1 and 2 won't work for sure
3 will work. but you have to take the whole zip file

thynkgit, 2024-05-24:
@mringwal Thank you for posting what you tried. Could you please post the exact URL that you used in the browser? I know that the IP address will be different, but knowing the proper format will be very helpful. Any other info you could provide about your success will be helpful as well. 😊 Thanks

markirb, 2024-05-25:
> @markirb Thanks a lot! That worked as expected. Can I help to document this? I would suggest to add a sentence that explains to use the zip file for the firmware upload. macOS Safari does unzip the archive in the download folder by default, that's why I didn't try uploading the .zip file.

Mhh I think the most troubling behavior here is Safari.... Maybe a note in the release section like "For Safari Please right click to download whole zip" would be enough.

This is for the beta anyway, as soon as there's a release this problem will go away.

thynkgit, 2024-05-25:
@mringwal Using the file upload function was the very first thing that I tried. That was where my first failed attempt occurred. I replaced the IP address as well. I performed it after upgrading Shelly from the original version to the latest version which is version 1.0 for my Plus 1. I will try it again. 

Question... what is the purpose of having "rojer" in the URL? From what I see that is the username and domain of another user.

thynkgit, 2024-06-05:
Thank you Mark! That gives me a better understanding of things. I tried the web UI firmware function again after my last post using a different URL. But now I will try it again using the URL that incorporates "rojer.me".

thynkgit, 2024-07-05:
@markirb Hi Markus. I'm back at it again and still with no luck. I have tried using the posted file in the browser and web ui. I do not understand what I am doing wrong. 

Also, I noticed that I called you "Mark" before, I apologize for that.

markirb, 2024-07-06:
Cannot help you without detailed steps what you really do and which files you use for which device.

thynkgit, 2024-07-31:
@markirb Hi Markus. I have some details for you as far as the URL I was using, and where i placed it. I have two pictures to add to my comments. 

This first picture is where I placed the URL in the Shelly web UI firmware section.

[image]

The next image is one of the browser response after putting the Shelly HomeKit URL in Safari's address bar:

[image]

I hope this information helps. Please let me know if you need any more details. Thank you Markus. Looking forward to your response!

thynkgit, 2024-07-31:
@markirb Markus, I just wanted to add one more note. After executing the URL in the browser, it just sat there and no further action was noticed. Thanks. Scott

markirb, 2024-07-31:
Update to latest (1.3.X) Shelly firmware first please.

closed 2024-08-29 by markirb