mongoose-os-apps/shelly-homekit #1393: Shelly Plus Plug S V2 Wifi-Issue

DB2FM, 2024-06-11:
I tried 2.12.0 Beta 3 on two Shelly PlusPlugS today. When disconnecting from mains and reconnecting WiFi is switched off. I had to connect to the AP and switch it back on. Settings were still there. 
Also the setting of LED doesn't change anything. It just doesn't light up.

markirb, 2024-06-11:
LEDs are not supported currently. It is only basic support. But thanks for reporting the WiFi issue.

Seems like it is not permanently saved somehow.
Can you maybe send logs?

DB2FM, 2024-06-12:
42202019810 shelly_debug.cpp:231    No log file, sending new entries
42202296214 shelly_main.cpp:473     Up 42202.29, HAP 0/0/16 ns 0, RAM: 208896/193748; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42210296712 shelly_main.cpp:473     Up 42210.29, HAP 0/0/16 ns 0, RAM: 208896/193748; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42218296472 shelly_main.cpp:473     Up 42218.29, HAP 0/0/16 ns 0, RAM: 208896/193748; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42225981599 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:56887
42226296749 shelly_main.cpp:473     Up 42226.29, HAP 0/0/16 ns 0, RAM: 208900/193748; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42226348274 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:56887
42226394648 mgos_http_server.c:180  0x3ffcdb2c HTTP connection from 192.168.22.44:52794
42226412243 mgos_mongoose.c:66      New heap free LWM: 193108
42226419240 mgos_mongoose.c:66      New heap free LWM: 189784
42226424504 mgos_mongoose.c:66      New heap free LWM: 189680
42226675835 mgos_http_server.c:180  0x3ffcdc2c HTTP connection from 192.168.22.44:52795
42226701555 mg_rpc.c:315            Shelly.GetInfo via WS_in 192.168.22.44:52795
42226748209 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:52795
42227834396 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:52795
42228710312 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:52795
42229954550 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:52795
42234296126 shelly_main.cpp:473     Up 42234.29, HAP 0/0/16 ns 0, RAM: 208700/189680; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42242296201 shelly_main.cpp:473     Up 42242.29, HAP 0/0/16 ns 0, RAM: 208912/189680; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0
42250296602 shelly_main.cpp:473     Up 42250.29, HAP 0/0/16 ns 0, RAM: 208912/189680; st 31; 0.1: st:0 in_st:-1 inm:-1 ininv:0

Does that help or should I do something different?

DB2FM, 2024-06-12:
Following some log-lines when saving Wifi-Info:

43328224881 mg_rpc.c:315            Shelly.SetWifiConfig via WS_in 192.168.22.44:52795
43328251568 shelly_wifi_config_:134 New config: {"ap": {"enable": false, "ssid": "ShellyPlusPlugS-E86BEAE41644", "pass": ""}, "sta": {"enable": true, "ssid": "JF-Office", "pass": "****************", "ip": "", "netmask": "", "gw": "", "nameserver": ""}, "sta1": {"enable": false, "ssid": "", "pass": "", "ip": "", "netmask": "", "gw": "", "nameserver": ""}, "sta_ps_mode": 2} 0 0
43328302999 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:52795

markirb, 2024-06-12:
> Following some log-lines when saving Wifi-Info: 43328224881 mg_rpc.c:315 Shelly.SetWifiConfig via WS_in 192.168.22.44:52795 43328251568 shelly_wifi_config_:134 New config: {"ap": {"enable": false, [...]

This is the correct approach, looks normal. However it should also be saved, and somehow I think it is not saved currently. Can you try to adjust some other setting. The log should read something like "conf9.json saved"

DB2FM, 2024-06-13:
> … <x-msg://18/#>

Looks good, so far (changed Output to inverted):

130122296735 shelly_main.cpp:473     Up 130122.29, HAP 0/0/16 ns 0, RAM: 208884/189680; st 29; 0.1: st:0 in_st:-1 inm:-1 ininv:0
130122309021 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:61854
130123938224 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:61854
130125473184 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.22.44:61854
130126096368 mg_rpc.c:315            Shelly.SetConfig via WS_in 192.168.22.44:61854
130126101931 shelly_rpc_service.:293 SetConfig ok, 1
130126218181 mgos_sys_config.c:323   Saved to conf9.json
130126224679 shelly_rpc_service.:296 Configuration change requires server restart
130126229318 shelly_main.cpp:338     === Destroying accessories
130126289140 shelly_main.cpp:279     === Creating accessories
130126294499 shelly_output.cpp:64    Output 1: on -> off (init)
130126301099 shelly_switch.cpp:259   Exporting 'Shelly Plus Plug S': type 1, state: 0

DB2FM, 2024-06-13:
The „Saved to config9.json“ doesn’t appear when saving Wifi-setup.

markirb, 2024-06-13:
Yes it does not save, that is desired behavior.

But it should as soon as you get an IP. 

This is to avoid wrong configs being saved. I tried to reproduce this on any other device but failed.... as soon as you get the IP it saved for me. But maybe there is still a bug that I cannot see, which is more obvious on plus devices

markirb, 2024-06-14:
I think I see the problem. The push button is not mapped correctly. So I think during startup the device thinks it needs to reset itself.
This will be fixed in next version

markirb, 2024-06-14:
please try beta4 

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta4

DB2FM, 2024-06-14:
Tried but now complete configuration is lost after un- and replugging.
Reboot is working without problems, when loosing power it’s reset to default.

markirb, 2024-06-14:
This is really strange... Complete configuration exactly sounds like reset... could be that the button is still inverted, but I do not see that...

markirb, 2024-06-16:
reset was inverted also, please try https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta5

DB2FM, 2024-06-17:
Hi Markus,

Now it works as expected :)

markirb, 2024-06-17:
Ok closing. LEDs will be a new feature

closed 2024-06-17 by markirb