mongoose-os-apps/shelly-homekit #1397: Issues encountered testing 2.12.0-beta4 on Shelly Plus 1 with Addon

BadgerBadgerAndFox, 2024-06-15:
Upgraded from latest stock to latest beta.
Unable to pair with Homekit if Mode is changed to Garage Door Opener - Homekit reports device is out of compliance
Reverting to Mode to Switch allows pairing to succeed.
Unable to downgrade to previous beta versions or release versions of the custom firmware. Error returned in UI: 
Error uploading: TypeError: Failed to fetch

Error reported in debug log:
517987209 mgos_ota_core.c:269     Starting, timeout 600, commit timeout 0, mem 196024
517996081 mgos_mongoose.c:66      New heap free LWM: 189976
518002603 mgos_mongoose.c:66      New heap free LWM: 189748
518019957 mgos_ota_core.c:656     FW: Plus1 esp32 2.11.2 20220812-225646/2.11.2-g1b146b4-master sig 0x00
518025689 shelly_main.cpp:360     == Stopping HAP service (1)
518030287 mgos_ota_core.c:771     begin: waiting for user
518035961 shelly_main.cpp:375     HAP server state: 2
518040579 mgos_mongoose.c:66      New heap free LWM: 189700
518049345 HAPPlatformTCPStrea:309 HAPPlatformTCPStreamClose ts 0x3ffcdbc8 nc 0x3ffce080 0/1/16
518057557 HAPPlatformTCPStrea:309 HAPPlatformTCPStreamClose ts 0x3ffcdd74 nc 0x3ffcdf40 0/0/16
518062405 mgos_mongoose.c:66      New heap free LWM: 188360
518075275 HAPPlatformTCPStrea:112 0x3ffc1634 0.0.0.0:9000 HAP connection closed, ns 0/0/16 i 1
518081070 shelly_main.cpp:375     HAP server state: 0
518085756 shelly_main.cpp:348     === Destroying accessories

Unable to revert to stock firmware. Apears to timeout and then waits for manual upload (the Gen 2 device firmware's are not publicly hosted so i'm not sure how to obtain the stock firmware to revert)

70540789 shelly_main.cpp:483     Up 170.53, HAP 0/0/16 ns 0, RAM: 208360/195376; st 28; 5.1: cur:open tgt:open cl:0 op:-1; 12.0: t:16.00; 12.1: t:81.90
170855529 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.236:54479
171681718 mgos_ota_core.c:269     Starting, timeout 600, commit timeout 0, mem 199176
171701830 mgos_ota_core.c:656     FW: Plus1 esp32 1.3.2 20240522-112837/1.3.2-g34c651b sig 0x00
171707529 shelly_main.cpp:360     == Stopping HAP service (1)
171712108 mgos_ota_core.c:771     begin: waiting for user
171717744 shelly_main.cpp:375     HAP server state: 2
171727179 HAPPlatformTCPStrea:112 0x3ffc1634 0.0.0.0:9000 HAP connection closed, ns 0/0/16 i 1
171732440 shelly_main.cpp:375     HAP server state: 0
171739643 shelly_main.cpp:348     === Destroying accessories
171744923 mgos_mongoose.c:66      New heap free LWM: 194824
171815673 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.1.236:54479

markirb, 2024-06-15:
Homekit reports device is out of compliance

-> this is normal as we have no certification
-> if pairing does not work it might also have to do with DHT sensor

The reversion/upgrade process is due to the DHT interval, as written in the other thread. Can be circumvented in beta by setting a large update interval. Next version version will fix this problem by halting sensor updates during upload. Might be necessary also during pairing

BadgerBadgerAndFox, 2024-06-15:
Thanks, yes i found the other thread related the changes introduced to introduce addon functionality. This allowed me to downgrade back to the current stable release.

I reverted back to beta4 and attempted to pair in GDO mode with the sensor interval set at 500 but encountered the same issue.

It attempts to pair as a bridge, which is different to the stable release which attempts to pair as garage door.

The beta only fails to pair in GDO mode. Switch mode works without issue. The debug log does not seem to capture anything of use during the pairing and reports as paired in the web ui even though it failed in homekit.

markirb, 2024-06-15:
Hi.

Yes the difference is the addon in your case.
So it will pair differently. As the device has multiple accessories it is now a bridge it can  - to my understanding - no longer be primary.

If HomeKit only supports GDOs as single primary then we just need to leave out the sensor...

markirb, 2024-06-15:
Ok I think the reason is that HomeKit does not allow Security related devices as door to be bridged. Will have to find another way

markirb, 2024-06-15:
Ok I tested it without bridge, and it works with sensors, so it will be fixed in next version; thanks for reporting

BadgerBadgerAndFox, 2024-06-15:
Fantastic, looking forward to testing the next release

markirb, 2024-06-16:
try https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta5

pedrodinis23, 2024-06-16:
Hi! Just flashed a Shellyplus1 with 2.12.0-beta5 with the addon and I have some issues: When in garage mode, it doesn't recognize the reed switch, so it gives me an Obstructed error when closing. When I cut the power to the shelly, in garage mode, it disconnects the wifi. Is there a way to fix this?
Thanks!

markirb, 2024-06-16:
Can it be that it now says you have an open sensor also? Can you deactivate it?

in comparison to which fw was this? Beta4 to beta5?

the closing logic has not changed…

for the WiFi there was nothing changed. I fear this is a general problem currently with any device which was always existing also in 2.11. but have to investigate

pedrodinis23, 2024-06-16:
Thanks for your quick reply, markirb!

The open sensor is deactivated. It would be great if I could deactivate the close sensor too!... The reed switch is working. I can see it on Switch mode. I tried yesterday with beta4 and today with beta5. The other thing is that it might work on Switch mode, if I could set an auto off with 500ms. But I turn Auto Off on, save it, and when I insert the amount of time and save, nothing happens. As soon as I flip the switch it turns on and off quickly and the time disappears.

markirb, 2024-06-16:
The open sensor is deactivated. It would be great if I could deactivate the close sensor too!...

=> it makes no sense to deactivate it. 
again, which version did this work? 2.11?

pedrodinis23, 2024-06-16:
It never worked correct. I flashed the Shelly with this link: http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update and it didn't work (I updated the shelly on the shelly app before this)... Then I found the fix, downgrade the shelly firmware and ran the link again. This second time I were able to get it on HomeKit firmware. I had the issue I told you about before, update to beta4 and then beta5. Same thing. 

Yes, I'm using the correct add-on. I have a reed switch attached to the bottom (ground and digital). It works on the shelly app and it works on HomeKit web interface on switch mode. When in garage mode, I doesn't seem to recognize the reed switch. It shows opened or stopped (obstructed), never closed. (and yes, I'm waiting a few seconds when it says closing and putting the magnet near the reed switch). It is not connected to the garage door yet. Just testing in the living room! :)

markirb, 2024-06-16:
Ah ok. Then it never ever worked with this firmware. This firmware has fixed closing sensor to the input of the Shelly. The addon can currently only work as opening sensor. So: not a bug but a feature request.

pedrodinis23, 2024-06-16:
Sorry… didn’t understand! You’re saying that a Shelly plus 1 with add on and a reed switch in garage mode doesn’t work at this moment?

markirb, 2024-06-16:
It does not closing sensor has to be connected to sw input. But careful: of course not on AC/230V. Wiring depends on your voltage and you reed. I have a DC setup that works just fine.

pedrodinis23, 2024-06-16:
Thanks markirb! Yes, I am aware of that. It wasn’t my first choice to install this, but I guess it will be the way to go. 👍🏼

pedrodinis23, 2024-06-16:
Just one more thing: what about the auto off function on switch mode? Is it working? I’m using 0:00:00:01:000 and it doesn’t save. Already tried to use other values, same thing.

BadgerBadgerAndFox, 2024-06-16:
> try https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.12.0-beta5

Great work!

This has resolved all the issues I initially encountered on beta4, only change i make post deploy from default is setting the second sensor and changing the update interval for the DHT22 to 5 seconds.

I need to check but i think my sensors are connected inverse to what is currently expected by the firmware. I'll check and adjust as needed.

markirb, 2024-06-17:
> I need to check but i think my sensors are connected inverse to what is currently expected by the firmware. I'll check and adjust as needed.
> 

Hm, I don'g get it. Are your sensor values swapped? Then your DHT provides swapped readings and the driver cannot deal with that. Do not know how to tackle this, what exact model do you have?

> May be worthwhile updating the diagrams to include Shelly with Addon https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings

Would be yes.

BadgerBadgerAndFox, 2024-06-17:
DHT22 sensor is fine. My open close inputs is where i had an issue.

Had my open close sensors connected the wrong way around (I was previously using native firmware and MQTT so the open and close sensors where whatever inputs i configured in the topics).

I incorrectly had the close sensor on the addon and open on the shelly. I have swapped them and the homekit accessory via your custom firmware is working correctly now.

closed 2024-06-17 by markirb