mongoose-os-apps/shelly-homekit #1400: 2.12.0-beta5 Seems to mess up my Shelly 2PM Wi-Fi connection. Beta 3 more stable.

hughfr4nc15, 2024-06-17:
Hello,

Just wanted to provide some feedback on this matter. I've been testing the Beta's on one of my "PM's and Beta 3 worked flawlessly. Beta 4 I believe was also OK but Beta 5 made the Wi-Fi very unstable/slow.
The device stopped responding very frequently and I was lucky to be able to roll back to Beta 3 which in my case seems the most stable.

Thank you.

markirb, 2024-06-17:
Thanks for reporting but nothing, really nothing in beta5 changed that could influence WiFi. With a lot of imagination it would be in beta4, but even then the relation is far fetched.

hughfr4nc15, 2024-06-17:
It might be also on beta4 because immediately after going to beta4 I seem to have noticed some instability but, when I checked a few minutes after was OK and I believe it stayed OK. On beta5 I updated last night and didn't check after... this morning shutter wouldn't open and after some check it seemed to be in some weird state where it was alternately reachable and unreachable in a matter of seconds. after some time It stayed reachable for a couple of minutes and I was able to downgrade to beta3 and it's working flawlessly now.

markirb, 2024-06-18:
Can you try if this one changes anything?

[shelly-homekit-ShellyPlus2PM.zip](https://github.com/user-attachments/files/15880668/shelly-homekit-ShellyPlus2PM.zip)

hughfr4nc15, 2024-06-18:
Hello @markirb,

Installed and running for 20m. Seems to be fine now.

Thanks.

closed 2024-06-18 by markirb