mongoose-os-apps/shelly-homekit #1401: +PlugS dead after unplugging and replugging

DockyAxl, 2024-06-17:
I updated to Shelly 2.12.0 beta 4. Everything worked fine, after unplugging and switching the outlet it didn't work anymore. Even the hardware button is not working. Unfortunately I confirmed the issue with a second +PlugS. :(

markirb, 2024-06-17:
I don’t think it’s dead. It just resets itself. Unfortunately reset button is inverted in beta4 version. So if you press it during plugging it it should work fine and come up as AP?

DockyAxl, 2024-06-17:
Thanks for the fast reply!
OK, I it seem to work if I press the button, but how can I get the device reconnected? It's not even in the network.

DockyAxl, 2024-06-17:
Thank you! I forgot to connect to 192.168.33.1 :) Issue closed

closed 2024-06-17 by markirb