mongoose-os-apps/shelly-homekit #1406: Shelly Plus1 update from 2.11.2. to 2.12.0 beta5 unfortunately not possible

campino60, 2024-06-24:
I have several Plus1 and also rgbw2 in use (HomeKit) and would like to update them.
unfortunately this is neither possible OTA nor via the web interface 
everything is loaded but the devices remain in rebooting status and do not continue.
after refreshing the page, the old FW version is still available
what am I doing wrong or what remedy is there?
thanks for the help

markirb, 2024-06-24:
Don’t know. Can you watch the logs while updating?

campino60, 2024-06-25:
here's the log file

[Update log shelly.txt](https://github.com/user-attachments/files/15964798/Update.log.shelly.txt)

markirb, 2024-06-25:
I do not see anything strange there.... seems like it is going right but afterwards reverts right away. You tried multiple times? Maybe try a factory reset and then again?

markirb, 2024-06-26:
```
52416609 [0;32mI (53059) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f91ch (194844) map[0m
52528419 [0;32mI (53169) esp_image: segment 1: paddr=0003f944 vaddr=3ffb0000 size=006d4h (  1748) [0m
52539172 [0;32mI (53179) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=cd9a8h (842152) map[0m
52995396 [0;32mI (53639) esp_image: segment 3: paddr=0010d9d0 vaddr=3ffb06d4 size=02f0ch ( 12044) [0m
53011220 [0;32mI (53649) esp_image: segment 4: paddr=001108e4 vaddr=40080000 size=143a0h ( 82848) [0m
53064423 [0;32mI (53709) esp_image: segment 5: paddr=00124c8c vaddr=50000000 size=00010h (    16) [0m
53072871 esp32xx_ota_backend:399 app_0 verified (d11ef68ca65eb11e64e3ca2b258896c8bd8c3715)
53081819 mgos_ota_core.c:820     Reached the end of archive
53089965 esp32xx_ota_backend:463 Setting boot partition to app_0
53099257 [0;32mI (53739) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f91ch (194844) map[0m
53215356 [0;32mI (53859) esp_image: segment 1: paddr=0003f944 vaddr=3ffb0000 size=006d4h (  1748) [0m
53226137 [0;32mI (53869) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=cd9a8h (842152) map[0m
53686185 [0;32mI (54329) esp_image: segment 3: paddr=0010d9d0 vaddr=3ffb06d4 size=02f0ch ( 12044) [0m
53702091 [0;32mI (54339) esp_image: segment 4: paddr=001108e4 vaddr=40080000 size=143a0h ( 82848) [0m
53755207 [0;32mI (54399) esp_image: segment 5: paddr=00124c8c vaddr=50000000 size=00010h (    16) [0m
53764491 [0;32mI (54409) esp_ota_ops: New OTA data 1: seq 0x00000029, st 0x10, CRC 0x6d7b3e02[0m
53844611 mgos_ota_core.c:1051    Update finished, result 1 (Update applied, rebooting)
53850812 mgos_ota_core.c:1202    Update state: 0
53855732 mgos_ota_core.c:1096    Update requires reboot
```

Actually the app_0 verified is missing for your case, maybe this is the reason?

campino60, 2024-07-10:
Actually the app_0 verified is missing for your case, maybe this is the reason?

What is the app_0 
i have the same issue with the Version 2.12.1
updating and the doesn't reboot

campino60, 2024-07-10:
is it necessary to uninstall HOMEKIT in order to perform an update?
Or can the Shelly be updated integrated in HOMEKIT?

markirb, 2024-07-10:
There is actually no requirement. But please try to revert to original Shelly and then flash again or try to do a factory reset. Nobody has this problem so it seems a problem of your device

campino60, 2024-07-11:
Many thanks for the help
there were problems with a shellyplus1
Update after revert to stock then also went through
all others (ShellyRGBW2, Plus1) could be updated without problems

closed 2024-07-11 by markirb