mongoose-os-apps/shelly-homekit #1407: Support for Shelly Plus RGBW PM

PabloIT, 2024-06-26:
Will the new Shelly Plus RGBW PM be supported by Mongoose?
https://www.shelly.com/en/products/shop/shelly-plus-rgbw-pm

markirb, 2024-06-26:
Oh wow did not know these exist.

If you or someone else here is up to donate the Hardware I would give it a try.

nerdosity, 2024-07-17:
Got 2x. How can I help? 

[image]
[image]
[image]
[image]
[image]

markirb, 2024-07-18:
Actually the pinout of the ESP32 would be needed. On the usual sites there is nothing yet...

nerdosity, 2024-07-18:
> Actually the pinout of the ESP32 would be needed. On the usual sites there is nothing yet...

From left to right, screw adapters on the downside:
+ GND GPIO36 GPIO37 GPIO38 GPIO39 GPIO25 GPIO26 GPIO27 GPIO24
So should be Input 1 to 4, R,G,B,W.

I did find everything with my multimeter.

Cannot find the onboard button yet, need a better magnifying lens.

markirb, 2024-07-18:
Oh wow very nice. I can start to work with that.... 

I can compile rpc-service-gpio (https://mongoose-os.com/docs/mongoose-os/api/rpc/rpc-service-gpio.md) in for the first version then you could check on the device if you want.

markirb, 2024-07-20:
Can you try to flash this? (cannot guarantee that it works, but normally this should not brick anything...)

Outputs should work like you expect, if you measured correctly :)

[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/16319568/shelly-homekit-ShellyPlusRGBWPM.zip)

You can also play around with GPIOs like this:

curl 192.168.188.56/rpc/GPIO.Write -d '{"pin":1,"value":1}'

curl 192.168.188.56/rpc/GPIO.Read-d '{"pin":1}'

https://mongoose-os.com/docs/mongoose-os/api/rpc/rpc-service-gpio.md

this way you can press reset button and see what happens or try to find the LED output

nerdosity, 2024-07-20:
I Will try, it's flashable via the UI?

markirb, 2024-07-20:
Yes it should be.

PabloIT, 2024-08-16:
@nerdosity Did you manage to test something?

lanmarc77, 2024-08-19:
Is someone able to determine what the maximum input frequency is that could be measured by the inputs? I assume the inputs are using a high pass filter at least it looks like it based from the pictures.

markirb, 2024-08-23:
@nerdosity any news?

@lanmarc77 do not know, what is the use case?

lanmarc77, 2024-08-23:
@markirb I actually would not see a LED strip controller but a FAN controller. But that would require the inputs to be reasonable fast to read the FAN ticks to allow PI(D) control of the FAN speed.

lanmarc77, 2024-08-24:
I can now answer the questions myself. I empirically determined that at around 300Hz (50% duty cycle) the rc filter cuts off. Removing a capacitor from one input I was able to detect until 15kHz and did not test further. 
So with this Shelly model FAN control is possible. For slower FANs without HW modification for faster FANs removal of the capacitor is needed.

lanmarc77, 2024-08-25:
I measured the following pinout which almost matches @nerdosity 
    IN1: GPIO36
    IN2: GPIO37
    IN3: GPIO38
    IN4: GPIO39
    R: GPIO25
    G: GPIO26
    B: GPIO27
    W: GPIO4 (it is pin 24 not GPIO24)
    Button: GPIO22
    LED: GPIO14
    Power measurement: GPIO35 (OUT of unknown opamp circuit from measuring R001)
    Voltage monitor: GPIO34 (via unkown resistor network)

And a picture of the PCB back:
[image]

markirb, 2024-08-25:
Ok nice work. I will try to integrate this during the week and give you a test build.

lanmarc77, 2024-08-26:
I have completely overwritten the Shelly. So I need a full setup with all initial ota, partiton table, bootloader....
This is what the Shelly partition layout looked like when I initially connected the serial adapter:
```
I (28) boot: Shelly OS loader 1.0.0 (Feb 25 2024 16:37:33 IDF 4.4.4-a8)
I (28) boot: chip revision: v3.1
I (29) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table @ 0x8000 (8ef27910):
I (60) boot: ## Label            Usage          Type ST Offset   Length   F
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000 0
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000 1
I (83) boot:  2 app_0            OTA app          00 10 00010000 00190000 1
I (91) boot:  3 fs_0             FS               01 82 001a0000 00060000 1
I (99) boot:  4 app_1            OTA app          00 11 00200000 00190000 1
I (106) boot:  5 fs_1             FS               01 82 00390000 00060000 1
I (114) boot:  6 aux              unknown          55 00 003f0000 0000c000 0
I (122) boot:  7 shelly           Unknown data     01 88 003fc000 00004000 1
I (130) boot: End of partition table
I (134) ota: Read BS0: seq 14 bst 0x11 seq2 14 | as 1 rs 0 ba 2 mfs 0 c 1 | v 1
I (142) ota: Read BS1: seq 14 bst 0x11 seq2 14 | as 1 rs 0 ba 2 mfs 0 c 1 | v 1
I (150) ota: Booting app 1
I (153) boot_comm: chip revision: 3, min. application chip revision: 0
I (160) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=2e75ch (190300) map
I (226) esp_image: segment 1: paddr=0022e784 vaddr=3ff80000 size=00004h (     4) load
I (227) esp_image: segment 2: paddr=0022e790 vaddr=3ffbdb60 size=01888h (  6280) load
I (234) esp_image: segment 3: paddr=00230020 vaddr=400d0020 size=13f624h (1308196) map
I (636) esp_image: segment 4: paddr=0036f64c vaddr=3ffbf3e8 size=0214ch (  8524) load
I (639) esp_image: segment 5: paddr=003717a0 vaddr=40080000 size=1d818h (120856) load
I (698) boot: Loaded app from partition at offset 0x200000
I (698) boot: Disabling RNG early entropy source...
cpu_start: Unicore app
```

markirb, 2024-08-26:
Try this:

[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/16747269/shelly-homekit-ShellyPlusRGBWPM.zip)

can you give me feedback then if everything works?
especially if the configuration is retained which means the button is correctly mapped and not inverted.

you have to flash via mos-tool

lanmarc77, 2024-08-26:
The back button is inverted. When I press it the serial console reads zero if I release 1, followed by a reset procedure.
Setting the color values results in core dumps. I attached one for you to check. I think at least red channel works as far as I could test.
Onboard LED seems to be ok. It is on during bootup then switches off. If I keep the backbutton pressed it flashes.
How would I test the inputs?

[core-PlusRGBWPM-esp32-20240826-134901.594212025.zip](https://github.com/user-attachments/files/16748189/core-PlusRGBWPM-esp32-20240826-134901.594212025.zip)

markirb, 2024-08-27:
> The back button is inverted. When I press it the serial console reads zero if I release 1, followed by a reset procedure.

This is expected behavior. So nothing to change

> Setting the color values results in core dumps. I attached one for you to check. 

Strange. I need some time to analyze this. I think it would be best to test in 4White mode so you can test each individual channel.

> I think at least red channel works as far as I could test. 

OK

> Onboard LED seems to be ok. It is on during bootup then switches off. If I keep the backbutton pressed it flashes. How would I test the inputs?

Good question. You can via above mentioned GPIO-RPC for example.
Otherwise, currently they are not really connected in the firmware :/
In 4 white mode we could have one switch for each white? It is not currently the case

> 
> [core-PlusRGBWPM-esp32-20240826-134901.594212025.zip](https://github.com/user-attachments/files/16748189/core-PlusRGBWPM-esp32-20240826-134901.594212025.zip)

markirb, 2024-08-27:
It should be? Maybe Pin 24 all along?

lanmarc77, 2024-08-27:
I meant how do I actually set it up? The webinterface only offers setting rgb but this model has an additional white channel.

lanmarc77, 2024-08-27:
And I am pretty certain it is GPIO4=PIN24. I marked the signal path from the ESP32 through the gate driver to the corresponding mosfet and inserted the relevant pinouts from the datasheet:
[image]

markirb, 2024-08-28:
> I meant how do I actually set it up? The webinterface only offers setting rgb but this model has an additional white channel.

[image]

lanmarc77, 2024-08-28:
Ah Ok. Thats how it looks on my device:
[image]

no Mode dropdown.

markirb, 2024-08-28:
Ah. My bad. Will send you a new version

markirb, 2024-08-28:
[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/16787900/shelly-homekit-ShellyPlusRGBWPM.zip)

this should work

lanmarc77, 2024-08-28:
That worked. White is outputting data and changes it according to what I see in the serial log. I guess we are done then?