mongoose-os-apps/shelly-homekit #1409: Unable to reset AP on shellycolorbulb

karthikiyengar, 2024-07-01:
I am experiencing an issue with one of my four Shelly Color Bulbs. The access point (AP) for this particular bulb appears to be password-locked, preventing me from connecting to it and configuring the device. It is possible that I might have misconfigured this bulb by accidentally setting a password: I don't remember as it's been a while. The other three bulbs are functioning perfectly and have been successfully paired with HomeKit.

I have attempted to follow the recovery steps, which involve power cycling the bulb, but I have been unsuccessful in resetting this specific bulb back to its factory AP settings. Furthermore, I am unable to pair the locked bulb using the Shelly app.

The AP names of the other bulbs follow the format "shellycolorbulb-XXXXXX". However, the problematic bulb's AP name is in the format "shellycolorbulb-XXXXXXXXXXXX", which might give us a clue on what went wrong.

I would greatly appreciate any guidance or suggestions on how to recover this potentially misconfigured bulb and restore it to its factory settings. Is there a way to salvage this bulb, or is it permanently locked due to the misconfiguration?

timoschilling, 2024-07-03:
> Furthermore, I am unable to pair the locked bulb using the Shelly app.

If you flashed the bulb with shelly-homekit, you can't use the shelly app anymore.

`shellycolorbulb-XXXXXX` vs `shellycolorbulb-XXXXXXXXXXXX` different versions of the shelly firmware have set different device names (which are used for the said). That has nothing to say.

You can reset them by turn the power off, turn them on, turn off after 3-9 sec, turn on again, repeat, after the 5 turn on it should be reseted.

karthikiyengar, 2024-07-03:
Thank you. The missing part in my reset process was the 3-9 second wait. Managed to reset it :-)

closed 2024-07-03 by karthikiyengar