mongoose-os-apps/shelly-homekit #1411: Shelly RGBW2 - change icon/function?

HannesHomeAss, 2024-07-06:
Hi

I use RGBW2 for irrigation. each channel controls its own valve. is it possible to change the icon or function so HomeKit understands that it is irrigation? Thanks!

markirb, 2024-07-06:
Hi,

are you using them as binary in/ouput?

Currently not possible, though it would not be hard to implement:
Add a new device mode: 4 Output with stateless switches, and then you could do exactly that.

HannesHomeAss, 2024-07-06:
Not sure what you mean with binary?

yeah it would be great if this feature could be implemented just like you can choose garage door with the Shelly Plus 1 when flashing

HannesHomeAss, 2024-07-06:
> Hi,
> 
> are you using them as binary in/ouput?
> 

Aah like on/off. Yes, I do. The valves I control
Is 24V. So only on or off. Even tough I could dim for the moment in HomeKit.

lucassite120, 2024-07-06:
You could do it with HAA firmware:
https://github.com/RavenSystem/esp-homekit-devices/wiki/water-valve

markirb, 2024-07-09:
https://github.com/mongoose-os-apps/shelly-homekit/tree/rgbw2_switch

I tried to build this quickly.
Please test:
[release-ShellyRGBW2.zip](https://github.com/user-attachments/files/16140361/release-ShellyRGBW2.zip)

HannesHomeAss, 2024-07-10:
Hi

How do I update with this zip OTA ?
Usually I just use the URL.

Thanks!

HannesHomeAss, 2024-07-10:
Tried:

￼

HannesHomeAss, 2024-07-10:
Think I solved it, but new error:
￼

HannesHomeAss, 2024-07-10:
[image]

markirb, 2024-07-10:
Oh sorry the zip is within the zip

HannesHomeAss, 2024-07-10:
Yes. Tried the zip in the zip.

markirb, 2024-07-10:
I see. Indeed the firmware is now too big... I have to see where I can take cuts

markirb, 2024-07-21:
Can you try this here?

You have to set it to "Normal mode"
[shelly-rgbw2-switch.zip](https://github.com/user-attachments/files/16324503/shelly-rgbw2-switch.zip)