mongoose-os-apps/shelly-homekit #1412: Factory reset - can't add the device again?

HannesHomeAss, 2024-07-07:
Hi

Had to do a factory reset for one of my RGBW2. After factory reset I can't flash for HomeKit again.
And in current firmware from Shelly you cant choose to upgrade from file?

Any ide how to solve this?

Thanks!

markirb, 2024-07-07:
Please see remark on main README for flashing from recent stock versions (you need to use current beta)

closed 2024-07-07 by markirb

HannesHomeAss, 2024-07-07:
Hi,

Use, I have tried with BETA too, but same issue.
How do i upload the current BETA when I cant choose any file within shelly GUI?

HannesHomeAss, 2024-07-07:
[image]

HannesHomeAss, 2024-07-07:
Found the issue, the device needs to be changed to colour mode in stock firmware, before you can flash to HomeKit.