mongoose-os-apps/shelly-homekit #1414: Shelly 2PM and AddsOn does not detect DHT22

hejsiri, 2024-07-08:
When AddOn is connected to Shelly 2PM, it does not detect the DHT22 sensor. In Shelly 1PM everything is fine. Temetaruta and humidity is shown in web gui and HomeKit.

markirb, 2024-07-09:
Hi,

I think you posted this in another thread. Can you try to give more information?

The auto detection of DHT sensors is quite tricky...

hejsiri, 2024-07-22:
Basically nothing more to say. Shelly 2PM + Addon + DHT22 does not work.

markirb, 2024-07-22:
There are several flavors of DHT sensors on the market. Some seem to work with the current solution some apparently do not.

Can you please post details about your model?

hejsiri, 2024-07-23:
AM2302

markirb, 2024-08-29:
I think this is related to the other issue with Addon not working. I will try myself when I find time.

markirb, 2024-09-08:
Hi, 

In think I found it.
Please look here:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1439#issuecomment-2336710479

closed 2024-09-08 by markirb