mongoose-os-apps/shelly-homekit #1416: Shelly RGBW2 - long push of input button for dimming function

lalolerry, 2024-07-12:
First of all I want to thank with deeply gratitude [markirb], [timoschilling] and all the contributors for the restart of this project and the long time waited new releases of this wonderful firmware, that I really love and use intensively on my shelly Devices. 

Then,
I have a small request/suggestion of function enanchment of the dimming function of Shelly RGBW2: 
I'm using my Shelly RGBW2 Homekit device in white mode, momentary input mode (connected to a push button), and it works like a charme. 
But unlike the original Shelly firmware, the push button isn't (also) used to change the brightness of the led stripes: 
short press to toggle on/off, **long press to step up/step down the brightness of led stripes**. So each time I decide to change brightness (I used to during the day/night) I have to take my phone out of the pocket and set it by apple home app, that is a little bit annoying. Maybe it's possible to add this function into the homekit firmware?
And a second function that is present in original firmware is that the external input controls the state of all channels, not only the first one. It's it possible to add the possibility to **choose if/which/all channel have to be driven by the input**?

I hope I was clear enought in my request.
Best Regards

Lalo

markirb, 2024-07-17:
Hi,

I think your ideas are very clear, and also reasonable.
Just cannot promise when any of them can be realized

lalolerry, 2024-07-17:
Ok Markirb, I’m glad to hear you also consider these as good enanchements and yes, of course, no hurry at all. 
Best Regards

Lalo

timoschilling, 2024-09-03:
double of https://github.com/mongoose-os-apps/shelly-homekit/discussions/1486

closed 2024-09-03 by timoschilling