mongoose-os-apps/shelly-homekit #1417: 2.12.1 Adding Input2 to Shelly 1PM

bmpalmeida, 2024-07-13:
Hi,

I just updated all my Shellies to this new version, and some of the Shelly 1PM, created an additional input. Is this as intended?

[image]

bmpalmeida, 2024-07-14:
I just discover another strange thing. I’m using iOS 18 beta. And the Shelly devices are appear duplicated. With one of them not responding. I already remove and add again. In the beginning they are only one but with time the second appears. 

[image]

markirb, 2024-07-14:
Is this the same device? The second input comes from addon detection. Something goes wrong here or do you have an addon?

I think it is shaky and then primary/non primary switchingleads to what you see.

bmpalmeida, 2024-07-14:
I don’t have any add on on this ones. I have around 20 1PM and all except one is showing the input2. The only one that doesn’t shows the input2 actually doesn’t have anything connected to the sw. 

Sorry for the screenshot. Indeed in the images they are not the same. But same behaviour on both.

markirb, 2024-07-20:
First:

Can you flash this please:

[shelly-homekit-Shelly1PM.zip](https://github.com/user-attachments/files/16319578/shelly-homekit-Shelly1PM.zip)

this should make the problem go away. Secondly, can you please tell me the output of:

curl [ip of device]/rpc/GPIO.Read-d '{"pin":3}'

bmpalmeida, 2024-07-22:
Yes. That version solve the problem. 

The output of the curl command is the following. 

{
  "value": 0
}

markirb, 2024-07-24:
Ok thanks for your input. 
That's what I expected. I will try to think of a better way for this device but have to try some things first

bmpalmeida, 2024-07-24:
If you need that I do some tests let me know.

jdesutter, 2025-01-06:
I am experiencing this exact issue on a shelly plus1 and could use some assistance removing the rogue input #2. I am a novice user and will need complete steps to fix this. Thanks Joe

jdesutter, 2025-01-07:
I tried to flash the above version on my Shelly Plus 1i4 and I get a message that says:

Host: http://192.168.1.10
Device Name: NorthFloodLight
Model: ShellyPlus1
Firmware: HomeKit 2.13.3
Incorrect Zip File for device...

Please advise if I am using the correct file for my Shelly device.

markirb, 2025-01-07:
When you're on Mac the zip file auto extracts. To avoid this use secondary click / right click download file as.

markirb, 2025-01-07:
Also this file is for 1PM. Which devices is it that fails for you?

jdesutter, 2025-01-07:
Markus,I have the Shelly plus 1 i4. I found a different file through the forum that wanted to upload, but it would stop saying the file was too large. If you could email the correct file id greatly appreciate it. JoeSent from my iPhone

jdesutter, 2025-01-07:
Yes I did this but my Shelly doesn’t like that file. Believe it is incorrect for the Plus1 i4 model I have. Thank you,JoeSent from my iPhone

jdesutter, 2025-01-07:
More info. I eventually was able to load newer versions of the Plus1 firmware. I uploaded  2.13.2 and 2.13.0. In both cases the input #2 was gone but reappeared. Once it stayed gone for awhile until i opened up home assistant, which i use for other devices. I'm wondering if it is triggering something in this Shelly.

markirb, 2025-01-07:
Ok so again: which device is this? You one time wrote Plus1? one time i4? what is it now?

Then I can try to work around this bug...

jdesutter, 2025-01-07:
It’s a plus1. Sorry for the confusion. Found the right file on your repository but it fail to up load because it’s too big. 2.13.0, 2.13.1 etc will upload and flash but seem to be much smaller files. Unfortunately they don’t fix my problem. Sent from my iPhone

markirb, 2025-01-07:
ok, then please read the upgrade notes here:
https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.13.3
this will explain it.

jdesutter, 2025-01-07:
Ok so if I read this correct. I have to revert to 1.3.1 stock, then update stock to like 1.4.0, or whatever is current, then I can upload the 2.12.1 update? And what does this mean? “Please still only do this if you accept the risk to loose your config in case you need to reset”Whatever I do I don’t want to brick it or have to pull it out of the wall. Anything else I can handle

https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.13.3

jdesutter, 2025-01-13:
Could someone please provide instructions to update my Shelly Plus1 to 2.12.1? I have reverted my relay to stock, but have not figured out how to flash anything other than the latest OTA update.

markirb, 2025-01-13:
Just follow the linked guide for OTA to latest 2.13. When this is done you can use webinterface to upload any zip file.

jdesutter, 2025-01-13:
Markus,

The problem is that it the web interface won’t allow a file older than 2.13.0. I can upload 2.13.0, 2.13.1 etc. I’m trying to fix the input#2 error mentioned in #1417 and according to your earlier work i need to use version 2.12.1.

Look i can tell you are very busy, but i’m trying to go backwards to 2.12.1 and I suppose because the upgrade path has been broken what you are saying below doesn’t work.

This release deliberately breaks the upgrade path from Shelly HomeKit 2.12.X for any Plus Devices (those are: +1, +1PM, +i4, +2PM, +RGBWPM, +PlugS). The reason for this is to support direct upgrade from newest stock firmware (1.4.X), that uses a changed partition layout.

Thank you,
Joe

pfrasun, 2025-01-20:
HI Markus,
Sorry I missed that thread. What do you mean it reboots, you mean self reboot? Crash, or also UI initiated reboot?
Looking at the runtime shown in the UI nothing indicated that Shelly was restarted, it looked like running uninterrupted for weeks and the duplicated input was observed during last days.
Thanks for all your help...

CyberMrProper, 2025-02-06:
@markirb this issue of second input is happening also for Shelly Plus 1. When the device restarts, or there is a power cut, then a second "dummy" device (besides having my principal as light) appears as a stateless button :O.

[image]

WRTHacker, 2025-08-28:
Hello experts,

is there any progress in fixing this issue?

Thanks and best regards

WRTHacker, 2025-08-29:
Hi Markus,
it's an shelly1+PM. If you need any further details, tell me how to get them!
Yes it's producable if I reboot the device, or reboot the main HomeKit controller (actually an AppleTV latest version).

A shelly1PM with 1.14.2 didn't show this behaviour!

WRTHacker, 2025-10-23:
Doesen't matter, thank you very much to try fixing it!
I fetch the 1.15.0-alpha0 and flash my Shelly Plus 1PM ... will observe the next days ... reboot Shelly and HomeKit Controller and will see if it stable.